EdgeTX/edgetx #368: Receiver number popup list

Mantamatt1, 2021-06-28:
Tx16s
Rc3
Relating to #367  can we look into having a list pop up showing which receiver numbers are in use so you can choose an empty slot rather than blindly trying different numbers to find one that is not in use. It gets frustrating when you have lots of models and no particular receiver order.

Maybe showing which models have which number so you can model match if you want to.

Example. Tap receiver number and a list appears for you to choose from.
If you select an unused slot that number becomes in use for that receiver/model.
If you choose a model already assigned with a receiver then a warning message is displayed warning that that receiver number in in use with that model and weather you want to continue!

Sorry for so many suggestions, but edge tx is quoted to be cutting edge! 😁

Imho open tx was way too clunky on colour lcd
Already edge tx is more fefined.
Stella job guys and gals

pfeerick, 2021-06-29:
> Sorry for so many suggestions, but edge tx is quoted to be cutting edge!

Don't be sorry, more ideas for the melting pot. Just don't get too close to the edge or you might get many cuts! 😁 

This would be a great idea, and is one I know that the OTX were actually going "unfeasible with the current design" ... so it's just a matter of figuring out what needs to be changed to make it happen... as part of the goal is to make it work better, not make your life harder! ;)

pfeerick, 2021-09-22:
I'm just adding to this again, as I see https://github.com/opentx/opentx/issues/8299 is about to close, and there was a little bit of discussion about *how* this could look there also...

Mantamatt1, 2021-09-22:
That's awesome. Look forward to it.👍

gagarinlg, 2023-03-18:
The warning part is already done for Color LCD. I am not sure about B&W.
What is missing is the drop-down with available receiver numbers, maybe color coded to show which ones are still free.