EdgeTX/edgetx #369: instant trim feature

pds59, 2021-06-28:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

pds59, 2021-06-28:
instant trim is set on momentary swich command looks like is working but no effect on actual trim

pds59, 2021-06-28:
firmware v2.4.0-rc3 radiomaster t16

closed 2021-06-28 by pds59

pfeerick, 2021-06-29:
I take it you've found it's been fixed (which is why https://github.com/EdgeTX/edgetx/issues/342 is closed) but a new nightly / RC with the fix wasn't out yet? It is working in the nightly that was build about two hours ago.

pds59, 2021-06-29:
Got It i try and did work the second time After i reset the functionTksPaolo de scaglia