EdgeTX/edgetx #372: Introduce 1+3, 2+3 and 3+3 widgets' layout

JimB40, 2021-06-29:
This will be particularly useful for Timer widget

pfeerick, 2021-08-23:
Since you mentioned it, did you mean 3x3 (9 zones), or 3 + 3 (six zones)?

JimB40, 2021-08-23:
Meant wwo colums  4 to 6 zones
Layouts naming is bit confusing now
Maybe we can name them better - eh this UX ;)

**1 column layouts**
Fullscreen -> 1 Col (1 Area)
not implemented -> 1 Column (2 Areas)
not implemented -> 1 Column (3 Areas)
not implemented -> 1 Column (4 Areas)

**2 columns layouts**
2 x 1 ->                      2 Column (1+1 Areas)
2 + 1 ->                      2 Column (1+2 Areas)
not implemented ->  2 Column (1+3 Areas)
not implemented ->  2 Column (1+4 Areas)
2 x 2 ->                      2 Column (2+2 Areas)
not implemented ->  2 Column (2+3 Areas)
not implemented ->  2 Column (2+4 Areas)
not implemented ->  2 Column (3+3 Areas)
not implemented ->  2 Column (3+4 Areas)
2 x 4->                        2 Column (4+4 Areas)

Choose what you want to implement ;)
however to my initial test those 1 column layouts: 1 Column (2 Areas), 1 Column (3 Areas), 1 Column (4 Areas) will work well for NV14

pfeerick, 2021-08-23:
I think Richard added "1 Column (4 Areas)" as 1x4. He's also added 1+ 2, which also 1 column, but is a big and then two smaller.

Yeah, naming is going to be a pain... 1CR2, 2C3R (column 1 has two rows, column 2 has three rows) 😆 ... just show me the picture! :)

JimB40, 2021-08-23:
Yep. if "2 Column (3+3 Areas)" is too long we may go with cryptic "2C 3+3A" there is always picture of widgets layout to confirm and dectypt :)

closed 2021-08-24 by pfeerick