EdgeTX/edgetx #374: Switch warning not active at model change

jogybaer0815, 2021-06-30:
EdgeTX nightly 2aa5296
TX16S

please have a look at the behavior: 
Switch warnings are only active when transmitter is switched on - a modelchange does not recognize 
preflight checks / Switch positions

in respect for your great work, Jörg

pfeerick, 2021-06-30:
Repeatable, TX16S on latest nightly. Also doesn't check the throttle state even if checkbox ticked. X9D+ correctly shows stick and switch warnings.

pfeerick, 2021-06-30:
May tie into implementing https://github.com/EdgeTX/edgetx/issues/348 also, since the end user experience was as if it had jumped out of the menu when selecting the model, show the pre-flightwarnings, then you'd be at the main screen/panel for the model.

raphaelcoeffic, 2021-11-22:
@pfeerick didn't we fix that one already?

pfeerick, 2021-11-22:
Yes, through #428. Thank you! :)

closed 2021-11-22 by pfeerick