EdgeTX/edgetx #375: Calculated Sensor Special function a<x max value 100mah

Ga1j1n, 2021-06-30:
I can create a calculated sensor for Consumption using Curr telemetry but if I create a special function to trigger an alert from that sensor the maximum value available is 100mah

Steps to reproduce the behaviour:
1) Add Telemetry sensors (in my case from betaflight)
2) Create a calculated sensor with Curr as the source and consumption as the type
3) Create a special function a<x with a mah value greater than 100mah (I use 1000mah as a low battery alarm trigger)

**Expected behavior**
In OtX any value can be assigned

**General information**
 - EdgeTX version (e.g. 2.4.0 latest
 - Hardware Jumper T18

pfeerick, 2021-07-01:
Reproducible on TX16S, not on X9D+.

raphaelcoeffic, 2021-07-01:
@Ga1j1n could you please show us the definition for the Curr sensor from BF? While testing here (no quad here, just the radio), I defined the Curr sensor by hand, and it seems the maximum value is dependent on the precision field in the current sensor itself. WIth precision set to "0.--", I can only to up to 100mAh. However, with "0.00" / unit "A" (which I think is the correct definition), I can set to a much higher value (I tried up to 16000mAh).

pfeerick, 2021-07-01:
Wait a minute... "0.--"... Does that mean after the decimal is ignored? Meaning it's Ah, not mAh, in the comparison?

pfeerick, 2021-07-01:
If it's the other way, shouldn't it go up to 999mAh?

pfeerick, 2021-07-02:
Actually, it's weirder than that... I hit the 100mah limit... went out of  the special function, went back in again, and now I can keep going... I stoppped when I got past 2500+

raphaelcoeffic, 2021-07-02:
> Wait a minute... "0.--"... Does that mean after the decimal is ignored? Meaning it's Ah, not mAh, in the comparison?

This setting only determines the exponent, not the mantisse. So basically, the raw value says for instance 9999, and then, based on the "precision", it will just shift the value. So "9999" if precision is "0.--", "999.9" if precision is "0.0" and "99.99" if precision is "0.00".

raphaelcoeffic, 2021-07-02:
> Actually, it's weirder than that... I hit the 100mah limit... went out of the logical switch edit page, went back in again, and now I can keep going... I stopped when I got past 2500+
> 
> If I use the VFAS (I've got Frsky voltage/current sensor on this model) telem object, same thing... I can go up to 1.00v, and no further. Exit and re-enter the logical switch edit page, and now I can keep going.
> 
> This line returns -100 for both the min and max values on the first run, and then appears to have correct values subsequent visits.

What is different between these two function calls? I believe the logical switch is modified in between, so that the function returns something different.

pfeerick, 2021-07-02:
Nope, for some reason appears to be the V2 range is never updated depending on what you actually set in V1... 

i.e. with my new friend TRACE and simu... the range updates when the edit page is built, when the func is chosen, but not when V1 is set ... and the V2 range is dependent on what is set in V1 😮 

https://user-images.githubusercontent.com/5500713/124237164-93508780-db5a-11eb-9edd-b9105e991a77.mp4

raphaelcoeffic, 2021-07-02:
> Nope, for some reason appears to be the V2 range is never updated depending on what you actually set in V1...

Ah ok, then this will be the issue. Should be easy to fix. Just check where V1 is modified, and trigger an update of the form from within these changes.

pfeerick, 2021-07-02:
I have a proof of concept fix for it now... basically hooked into where v1 invalidates v2 already, and if v2 exists, adjust it's min and max values... just waiting the firmware and simu to rebuild off the current 2.4 and test on the TX before I push the PR branch...

closed 2021-07-04 by pfeerick

Ga1j1n, 2021-09-13:
just an update,

I noticed an unusual scenario on 2.4 release Jumper T18, I added a Calculated sensor (mah consumed) derived from a FrSky Current sensor and the radio just freaked out with the selection highlight bouncing around on what I recall to be outputs (can't be 100%), I tried deleting the problem line but the issue would just move to another, eventually I realised that deleting the calculated sensor eliminated the behaviour.

Are you happy this is completely resolved as I tend to rely on these in conjunction with a logical switch to advise when I need to land rather than a basic timer for my larger electric models, great work overall BTW

PR #383 (2021-07-04): Reset logical switch v2 range if v1 changes
Resolves #375 

Use the v1 handler to update v2 range if the v2 source box has already
been created, so that the min/max values are correctly updated from v1.

Read it and weep over my atrocious code. :-P

As always, open to suggestions and being told there's a better way to do it. TX16S hasn't crashed yet :-P