EdgeTX/edgetx #376: Feature Request: SYS "Tools" Lua Script Page

CorruptName, 2021-06-30:
Hello and thanks for sharing your awesome work.
On the first page, when you hold SYS key, You get the Lua Scripts which are in the "SCRIPTS/TOOLS" folder.
For these Scripts, can we get larger buttons? Maybe square, the size of your finger? Then can we get assignable Images? Almost like an app drawer with "App Sized" Icons? Would be awesome.
Just an idea.
Thanks for considering.

raphaelcoeffic, 2021-07-01:
> Hello and thanks for sharing your awesome work.
> On the first page, when you hold SYS key, You get the Lua Scripts which are in the "SCRIPTS/TOOLS" folder.
> For these Scripts, can we get larger [...]

Thanks, this is already planned for next release. We will increase the size of quite some fields to make it more compatible with “bigger fingers” ;-)

JimB40, 2021-11-11:
This became more obvious with NV14 release.
In SD Browser or Tools lines are so tiny that error rate is like 60-70%

PR #1265 (2021-12-18): fixed 376 by making each line 10 pixels higher.
Resolves #376
closed 2021-12-18 by pfeerick