EdgeTX/edgetx #380: Internal GPS on Jumper T18 (or RadioKing TX18S)

ViorelX, 2021-07-01:
Hello,
I am trying to get working an internal GPS added to my radio on UART8. Initially, I posted this issue on the OpenTX forum: https://github.com/opentx/opentx/issues/8582#issue-932393915, but yesterday I have found this great new fork, EdgeTX.

What have I done so far?
I added some code lines on hal.h:

#if defined(RADIO_T18) && defined(INTERNAL_GPS)
#define GPS_RCC_AHB1Periph RCC_AHB1Periph_GPIOE
#define GPS_RCC_APB1Periph RCC_APB1Periph_UART8
#define GPS_RCC_APB2Periph 0
#define GPS_USART UART8
#define GPS_GPIO_AF GPIO_AF_UART8
#define GPS_USART_IRQn UART8_IRQn
#define GPS_USART_IRQHandler UART8_IRQHandler
#define GPS_UART_GPIO GPIOE
#define GPS_TX_GPIO_PIN GPIO_Pin_1 // PE.01
#define GPS_RX_GPIO_PIN GPIO_Pin_0 // PE.00
#define GPS_TX_GPIO_PinSource GPIO_PinSource1
#define GPS_RX_GPIO_PinSource GPIO_PinSource0
#elif !defined(INTERNAL_GPS)
#define GPS_RCC_AHB1Periph 0
#define GPS_RCC_APB1Periph 0
#define GPS_RCC_APB2Periph 0
#endif

These are similar to the original lines for setting up the internal GPS of Horus X12S, but instead UART4 I use UART8 and the TX, RX pins are changed accordingly.

Unfortunately, it doesn't work, neither in OpenTX, nor in EdgeTX. On power up, the screen becomes gray and the radio gets blocked.
What is wrong, what should be done?

raphaelcoeffic, 2021-07-01:
> On power up, the screen becomes grey and the radio gets blocked.

Could you please list:
- `cmake` parameters used when compiling
- other connections to other UARTs

I'm a bit puzzled as to why you chose to go for UART8, rather than any other. As for the issue on boot, there is probably some conflicts on peripherals, you will need to check things methodically. At best, you should try to enable as few things as possibly to try to isolate which combination is failing. Normally, initialising a UART should not cause the boot sequence to fail.

rotorman, 2021-07-01:
@ViorelX It would be more easy to follow your changes if you would:
1) fork EdgeTX
2) create a branch (e.g. T18GPS) in your fork
3) commit all your changes in that branch and
4) post a link to it for us to have a look.

ViorelX, 2021-07-02:
Thank you for the replies.
OK, I managed to do these tasks (right now, I am learning to work on github).
Here is my branch:
https://github.com/ViorelX/edgetx/tree/T18_GPS

Why have I chosen UART8 for the GPS?
As I know:
USART1 is in use for the internal module;
USART2 is for telemetry;
USART3 is for serial port (debug) and for R9M ACCESS;
UART4 needs the PA0 and PA1 pins as TX, RX terminals, but these are also used by the left stick;
UART5 needs PC12, PD2  but the parameter GPS_UART_GPIO can be set either to port C or D;
USART6 is for Bluetooth;
UART7 needs PE8, PE7 - I have tried this, but it doesn't work. Maybe, I will try again;
UART8 needs PE1, PE0 - this is what I am trying now.

ViorelX, 2021-07-03:
I have found the SDRAM driver uses a lot of CPU pins, as commented in this source file, lines 72-92:
https://github.com/EdgeTX/edgetx/blob/590ec5a9ed79bf1ed8fc8cb99601e37c7da0b0ae/radio/src/targets/horus/sdram_driver.c
These include PE0, PE1 needed by UART8 and PE7, PE8 needed by UART7.

Therefore, I reconsidere UART4 with the alternative use of PC10, PC11. No other use have I noticed so far. This is the last chance.

This morning I have updated that lines for setting up the GPS, compiled the project, flashed to the radio and ... IT WORKS!

Wait a little to change the wires of the GPS and I will announce if it really works.

ViorelX, 2021-07-03:
I have soldered the TX, RX pins of the GPS (Beitian BN-800) to the 161, 162 CPU pins (to use the UART4 terminals PC.10, PC.11).
It doesn't work.
At startup I get the message 'Bad radio data' and the SD card is no more accessible.
Otherwise, the radio works normally. On the 'statistics' page it appears the GPS, but it gets not fixes, even after a long run time.

I desoldered the RX, TX wires from the GPS and the SD-card works again.

pfeerick, 2021-07-03:
Hm... that sounds strongly like PC10 and /or PC11 are being used by the SD card... Do any of those pins have SPI as a secondary function... perhaps chip select?

ViorelX, 2021-07-03:
Yes, both of them have an alternate SPI3 function.

ViorelX, 2021-07-03:
UART5 is the only one port I have not tried. It uses PD2, PC12 (without alternatives), but these are by the SD card too.
Therefore, no more chance!!!

rotorman, 2021-07-03:
@ViorelX huh, I am a bit amazed that you are wildly hooking up your GPS to pins who's usage you are not clear of. I would suggest first thing to do is map all the used pins of T18/TX18S. This could be done purely from EdgeTX source, as I have done for TX16S as here:
[image]
Only then you can make better educated decisions where you can add additional components and where not.

A bit radical, but even better solution would be to fully trace all the tracks on the PCB, as e.g. done here by removing the solder mask and delaminating the layers:
[https://www.rcgroups.com/forums/showthread.php?3868769-Blog-16-RadioMaster-TX16S-mainboard](https://www.rcgroups.com/forums/showthread.php?3868769-Blog-16-RadioMaster-TX16S-mainboard)

pfeerick, 2021-07-03:
Ouch... well, that explains the SD card error! 😢 It sounds like you may not be able to have GPS unless something else like the bluetooth gets the boot!

brainbubblersbest, 2021-07-04:
I think thats the reason for the additional spi connected 12channel adc7952 IC @rotorman mentioned on that Radio. And why there is a free UART 4 for exclusive the GPS. Because less connected adc Lines on stm.
A Solution here could be changing the UART usage at runtime like  he said someday in the future. 🙂

There is also a Gyroscope/Accelerometer Board on the x12s that never had worked with Opentx and is complete useless. I Think it's connected to another uart/spi thats only available on that Radio.

rotorman, 2021-07-04:
@brainbubblersbest I believe you are confusing radios - I said X12S has Texas Instruments [ADS7952](https://www.ti.com/product/ADS7952).
Although I have not yet seen T18/TX18S myself, looking at the images found on the net, I do not believe it has external ADC.

brainbubblersbest, 2021-07-04:
Thats what i mean to say. He will connect gps+Bluetooth+eaMOD the same time. But All x10 based Radio processor Pins are already in Use. So the UART4 Pins are here used with alternate Function. (left gimbal?)

Only way to use more then two Devices would be with a connector and plug in the stuff he need for his used setup and Model. 
Then configure it in the Hardware Page. Like you add a BTModule to inav. But in exchange remove the Lidar.

rotorman, 2021-07-04:
SPI Flash is currently unused on OpenTX/EdgeTX on color radios, so that would be one option to go (e.g. with SPI to UART converter IC there).

brainbubblersbest, 2021-07-04:
I read about other Plans with the spi Flash for edgetx.
For that Single Radio this would be a solution. 🙃

rotorman, 2021-07-04:
Even with future usage of SPI, you only need a single dedicated ChipSelect line (any GPIO) for the UART converter.

ViorelX, 2021-07-04:
> @ViorelX huh, I am a bit amazed that you are wildly hooking up your GPS to pins who's usage you are not clear of.

@rotorman, that's because I am a simple hobbyst, not so much experienced. I am new to OpenTX, github and such radios. I have purchased a Radioking a few months ago, but due to lack of time, I didn't use it yet. But I strongly want to modify it.
I am also amazed to see your great work. I have studied your blog and now I am much more documented than yesterday. Thank you very much for all that stuff.
Now I know my radio has at least these unused pins: PA5, PA13, PA14, PA15, PH13.

Here is my last solution:
The left stick (gimbal) uses PA0, PA1.
I'll cut the gimbal wires connected to PA0, PA1 and move them to PA13, PA14, which belong to the (unused)  Serial-Wire-Debug (SWD).
Instead, I will connect the GPS TX, RX wires to PA0, PA1, which can be configured as TX, RX pins of UART4.
Is this a bad solution?

pfeerick, 2021-07-04:
@ViorelX I think part of the amazement is due to the fact that you _could_ let the magic smoke out of the radio by doing things that way... 😮 As long as you are aware of this risk... 😉

brainbubblersbest, 2021-07-04:
> > @ViorelX huh, I am a bit amazed that you are wildly hooking up your GPS to pins who's usage you are not clear of.
> 
> @rotorman, that's because I am a simple hobbyst, not so much experienced. I am new to OpenTX, github and such radios. I have purchased a Radioking a few months ago, but due to lack of time, I didn't use it yet. But I strongly want to modify it.
> I am also amazed to see your great work. I have studied your blog and now I am much better documented than yesterday. Thank you very much for all that stuff.
> Now I know my radio has at least these unused pins: PA5, PA13, PA14, PA15, PH13.
> 
> Here is my last solution:
> The left stick (gimbal) uses PA0, PA1.
> I'll cut the gimbal wires connected to PA0, PA1 and move them to PA13, PA14, which belong to the (unused) Serial-Wire-Debug (SWD).
> Instead, I will connect the GPS TX, RX wires to PA0, PA1, which can be configured as TX, RX pins of UART4.
> Is this a bad solution?

I really like this "lets Do it" mindset 😆
I would try it. Knowing something Bad can happen🤣
But no risk no fun 🥳

rotorman, 2021-07-05:
@ViorelX SWD is a awesome programming & in-circuit-debug channel to the µC/radio to see what is happening during execution. Personally, I would not sacrifice SWD and instead find other pins to be re-mapped. For more about in-circuit debugging plz see: [https://github.com/EdgeTX/edgetx/wiki/In-Circuit-Debugging-with-Qt-Creator](https://github.com/EdgeTX/edgetx/wiki/In-Circuit-Debugging-with-Qt-Creator)
As a modder, I think this tool will come very useful for you.

If you need further free pins, consider again SPI. Similarly as done in X12S, you could add an ADS7952 breakout to sample up to 12 of your analog inputs (the code is also there in https://github.com/EdgeTX/edgetx/blob/2.4/radio/src/targets/horus/x12s_adc_driver.cpp making the integration easy.

ViorelX, 2021-07-07:
I give up the GPS momentarily. There is no simple solution. My last idea to use the SWD pins as input pins for the left gimbal doesn't work: that pins cannot be configured as ADC inputs. Only certain pins may serve as inputs for the 3 ADCs with 16 channels and no more pins are available. Therefore, the SWD will be preserved as it is.
Eventually, could the SWD work with a GPS with I2C interface?
The use of an ADS7952, as done in X12S, would be a great solution, but at the moment, it is too complicated for me. It requires a lot of work and time.
Now I am looking to put a GPS with SPI interface on SPI2 (the unused SPI of the EPROM, i.e. PI0, PI1, PI2, PI3 pins). I need a GPS like this:
http://www.farnell.com/datasheets/1821035.pdf

rotorman, 2021-07-07:
> Eventually, could the SWD work with a GPS with I2C interface?

No, SWD is only for debugging, you cannot control it from firmware.

sprrrry, 2021-07-08:
I don't know much about all this but with the help of @rotorman's tx16s pcb diagram, I found one of the external uart of tx16s is connected to the bluetooth module pads in the pcb and the other is connected to the debug uart. Both of these uarts are broken out in the t18 pcb, so can't a gps be installed by connecting to the same uart as in tx16s and make similar code modifications? Please correct me if I'm wrong...

ViorelX, 2021-07-08:
No, because I have put a bluetooth module there (UART6). And at the debug port (UART2) I have made connections for R9M ACCESS.

brainbubblersbest, 2021-07-09:
If you dont have Touch, you can use 198 and 199.
Its used for gyroscope on the x12s. Bu not implemented with opentx or edgetx

brainbubblersbest, 2021-07-09:
[image]

ViorelX, 2021-07-09:
Thank you, but I have touchscreen.

Is this a good tool for SWD?
https://ardushop.ro/en/home/1288-programator-compatibil-st-link-v2-stm8-stm32.html
It's the cheapest I can get.

rotorman, 2021-07-09:
Have a look at Segger J-Link EDU Mini, the cheapest **original** for non-commercial purposes:
https://www.segger.com/products/debug-probes/j-link/models/j-link-edu-mini/

K2AAE, 2022-01-16:
> [image]

Now do this to the T18 Pro V2 mother board! I like the lighted switch labels of the T18 but not having access to uart for GPS or other functions means this unit will be replaced.

rotorman, 2022-01-16:
The easiest to get internal GPS on T18, would be to attach it to Bluetooth module pads.

[image]

ViorelX, 2022-01-18:
OK, but I already have mounted the FrSky Bluetooth module on its pads.
Actually, it doesn't work for me: on my phone it appears in the list of devices which can be connected by Bluetooth, but I didn't managed to pair it. I also tried to connect T18 with Inav Mission Planner, without success. The only way to connect it is via USB.

rotorman, 2022-01-18:
I have used FrSky BT module in RM TX16S successfully with the HeadTracker Pj. https://github.com/dlktdr/HeadTracker (basically Arduino Nano 33 BLE). It also pairs OK with another FrSky BT module.
Some info about the protocol used by the module: https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page662#post46878713

sdellava, 2023-08-06:
> have mounted the FrSky Bluetooth module on its pads.

Hi, I'm trying to install a GPS on a Jumper T18 Pro. Did you get it works?

sdellava, 2023-08-06:
> @K2AAE The easiest to get internal GPS on T18 IMO, would be to attach it to Bluetooth module pads:
> 
> [image]
> 
> I would look for alternative hookup options, only if you are planning to use FrSky Bluetooth module.

I've followed your suggestion, but nothing happen. GPS is 9600 bps and I've configured it on the AUX2. but nothing happen on the radio:

[image]
[image]

Brownsy67, 2025-03-20:
I know I'm late to the party here. but do you happen to know the max current output on the 3.3 pad of that spot is? trying to figure out if I can power an esp32 as BT module with it.