EdgeTX/edgetx #382: Unify the aux_serial compile options on all radios.

brainbubblersbest, 2021-07-02:
Please unify the Code for the aux Serials for All radios in the same way
1-On x10s / x12s the Bluetooth is not named as an Option in the make files and the hal leading to aux_serial2 is not available. 
So if compiling with flag -dbluetooth=OFF i can not use - daux_serial2=ON

2-On x10s express or x10s x12s with externalaccessmod aux_serial 1 is not available. This is ok.
But aux_serial1 is only available on debug builds on this Radios, even without express/access use of uart3

I will try ollis mav link integration on that Radios. 
Unifiyng the way how the Serials are handled should be changed to make this possible. 

A compile Option in future flasher releases like "BTISserialaux"  can be used to make a Serial available, by removing the bt module as a Opt In.

brainbubblersbest, 2021-07-02:
I just compiled a firmware for the x12 and x10 out of the mav-link branch. The cmake file just needs to changed from Set BLUETOOTH ON
TO
Option BLUETOOTH ON

Maybe i had better comment this in the mav-link branch authored by @raphaelcoeffic 😅 everything seems to work as it should 😁

raphaelcoeffic, 2021-07-02:
This is just one part of the job. The other part would to unify the code implementing the serial ports, as it is a huge heap of copy-paste right now, located at 3-4 different places right now....

brainbubblersbest, 2021-07-02:
As you done in the mav-link branch already. Or not? A year ago i searched for every change 3djc make to the Code to implementing the aux2_serial for the tx16s. I know this is not all it needs. Does it make sense to leave this open?

raphaelcoeffic, 2021-07-02:
> As you done in the mav-link branch already. Or not? A year ago i searched for every change 3djc make to the Code to implementing the aux2_serial for the tx16s. I know this is not all it needs. Does it make any sense to leave this open because this already planned for the mav-link branch?

I don’t mind either way.

brainbubblersbest, 2021-07-02:
I can use AUX2_SERIAL ON X12S with the additional change described above.
But AUX1_SERIAL seems to be Hidden in the UI outside an Debug Build. Where in the Code is this located.

brainbubblersbest, 2021-07-02:
[image]

brainbubblersbest, 2021-07-02:
[image]

brainbubblersbest, 2021-07-02:
It's working!
-DAUX1_SERIAL=YES Was wrong.
The 1 should not be there.
Testing the x12 with the mav_link branch now.