EdgeTX/edgetx #386: Timer Issue

Mikey1968, 2021-07-02:
The issue I'm seeing is if I go to count down . When I put it on 10s it's giving me 20s count down.

raphaelcoeffic, 2021-07-02:
> The issue I'm seeing is if I go to count down . When I put it on 10s it's giving me 20s count down.
> 
> Also I decided to try 5s and now it counts even higher. Not sure what's going on

I believe this one has been fixed a couple of days ago. Can you please check which version you are on?

Mikey1968, 2021-07-02:
I have the lastest version . I downloaded the firmware a few days ago.

raphaelcoeffic, 2021-07-02:
> I have the lastest version . I downloaded the firmware a few days ago.

Aha! A few days ago might mean a world difference right now. Could you please re-try with the latest nightly build?

Mikey1968, 2021-07-02:
Do you have the link

Mikey1968, 2021-07-02:
Where do you download those at.

TramALot, 2021-07-02:
The flasher

pfeerick, 2021-07-02:
If you're using the flasher, try the branch name ending in 'nightly'. Otherwise, check the 'nightly' release on the releases page. 

There was a fix done for this four days ago.

pfeerick, 2021-07-04:
@Mikey1968 Is this still an active issue, or can it now be closed?

Mikey1968, 2021-07-04:
I will give it a try today and let you know what happens

Mikey1968, 2021-07-04:
The problem was fixed. Thanks so much for the help

closed 2021-07-04 by rotorman