EdgeTX/edgetx #387: Modes active in Mixes - #8 cannot be de-selected

MartyFufkin, 2021-07-02:
I searched for Modes in Mixes in open issues and didn't see anything.

**Describe the bug**
When editing a Mixer channel, the all the modes can be selected or de-selected (yellow to light blue in default theme)

**To Reproduce**
Model Settings --> Mixes --> Select any channel and Edit --> Try to de-select Mode 8.  All other modes can be de-selected except number 8 (remains yellow).

**Expected behavior**
I expect number 8 to toggle like numbers 0 through 7.

**Screenshots**
[image]

**General information**
 - EdgeTX version 2.4.0 d878efb3
 - Hardware: Radiomaster TX16S

jogybaer0815, 2021-07-02:
TX16S nightly d878efb
confirm non desectability of Mode 8

pfeerick, 2021-07-03:
Yeah, confirmed here also. RM-TX16S, touchscreen fix PR firmware.

pfeerick, 2021-07-03:
@raphaelcoeffic This is the offending line
https://github.com/EdgeTX/edgetx/blob/4667b296989dc19c0d6915c031f401a6f28542d5/radio/src/gui/colorlcd/model_mixes.cpp#L145

as opposed to this one for the inputs page, which doesn't have a problem...
https://github.com/EdgeTX/edgetx/blob/4667b296989dc19c0d6915c031f401a6f28542d5/radio/src/gui/colorlcd/model_inputs.cpp#L293

**BUT** am I missing something, and they're BOTH wrong? Since the variable `i` has been declared as a `uint8_t`, yet, in the case of the admittedly apparently working inputs page (where a quick test with FM8 seemed to work on the transmitter), it it seems to be doing a bit manipulation treating it as a `uint32_t` ... but there's nine flight modes with a eight bit value?

I'm thinking a PR changing the i in both instances uint32_t across the board for that loop?

lshems, 2021-07-03:
Flightmode 0 is assigned when NO other flightmode is assigned. So perhaps when the parameter is null, it should default to 0, and when it has a value not null, it will get the value. Sneaky way to get nine bits out of a byte?

pfeerick, 2021-07-03:
I don't think so... it's a uint32 value in the underlying data structures, plus it's inconsistent between implementations. In the case of the mix page, the flight mode just can't be toggled, in the case of the inputs page, I got no idea why it actually seems to works, and with the headache I currently have, I'm in no mood to step through the logic problem. 😆 

Actually setting to uint32_t values does appear to work though... but will need more testing obviously...

lshems, 2021-07-03:
Ah, the headaches paradox. Lots of people are not in the 'mood' because of a headache. But doing some 'fun' stuff takes away blood from the parts in the brain causing the headache. So the general advice I always give: just have fun!

lshems, 2021-07-03:
So you had your pleasure fixing this one after all. Lol.

pfeerick, 2021-07-03:
I decided to play wack-a-mole and test it tomorrow ... if that fails, I'll delete it, deny it ever happened, and we won't ever speak of this issue again. 😆

pfeerick, 2021-07-04:
@MartyFufkin Would you mind giving #394 a try and see if it works properly for you... 

Direct link to the firmware files => https://github.com/EdgeTX/edgetx/suites/3149558893/artifacts/72406020

MartyFufkin, 2021-07-04:
I'm away from home next couple of days but I will try it Monday or Tuesday.

MartyFufkin, 2021-07-05:
I flashed the firmware from the Direct Link and I can confirm that this issue appears to be fixed.  Nice work guys.

PR #394 (2021-07-06): Use uint32_t values consistent with underlying data structures
Resolves #387 

For some reason, the input page appeared to work, but was mixing uint8_t and uint32_t data types... so I've updated both to align with the underlying uint32_t used by the actual data structures. 

Need to test this further on the TX as only simu tested to verify that the button actually toggled now - I haven't verified the flight mode is behaving correctly yet.
closed 2021-07-05 by MartyFufkin