EdgeTX/edgetx #388: More global variables

jfrickmann, 2021-07-02:
More global variables, please! I suggest that we increase the number from 9 to 16.

lshems, 2021-07-02:
At cost of what?

I don't know much about memory, but names of inputs, outputs, mixers could all be stored on the sdcard and loaded when needed. Same as for sensor names.

Then also store the name on the sdcard associated with the numbers to do inverse mapping of sensor names on the sensor numbers so you can remap after rediscovery.

pfeerick, 2021-07-03:
Plus also allowing for slightly longer names?

raphaelcoeffic, 2021-07-03:
> At cost of what?
>
> I don't know much about memory, but names of inputs, outputs, mixers could all be stored on the sdcard and loaded when needed. Same as for sensor names.
>
> Then also store the name [...]

Loading names on demand from SD card could be a huge pain. However, having longer names and more global bars on color lcd is not a problem. I would not touch the b/w however.

lshems, 2021-07-03:
Loading names on demand could be loading names on model load. I see lots of code being added.does that not take memory? Apparently not the memory that is so limited. So load it once? Just a noob here, no clue what I'm talking about.

pfeerick, 2021-07-03:
Indeed, that probably wouldn't be such an issue... as you're basically doing that anyway when you're loading the model from SD... you're loading all the various bits into RAM for use later.

jfrickmann, 2021-07-03:
Guys, let's not over complicate this. Just change MAX_GVARS from 9 to 16 in dataconstants.h and see what happens. If the tiny memories of Taranis radios cannot handle it, then we can either keep 9 gvars for those radios, or find another place to save memory.
E.g. remove the names from mixer lines. Yes, I know that will upset some people, but we also do not have names for logical switches, and we have to make compromises. I find it much more useful to have more gvars, and I have heard the same thing from Pat Mckenzie and Reto Fiolka, so I am not the only one!

lshems, 2021-07-03:
In my opinion we cannot go for a compromise, since mixer names are crucial in finding things for preset templates. They would at least kill all my model apps.

Meaning, I would love to have more gvars, but not at the cost of mixer names.

There are other places to save memory. The lua mixers page, that most people activate with the Lua option but do not need at all, is perhaps limiting the number of gvars.

In that case, make the extra gvars available if the lua mixers are not, and vice versa. 

Yes, compromise, but a better one than killing mixer names.

lshems, 2021-07-03:
Also, just for info, some things don't need gvars. You can work around.

I use two mixer lines to create a unique value for each FM, and then use a mixerlobe multiply with that input and a curve to adjust weights. One gvar less busy with what is mostly a one time setup: rates on surfaces per flightmode.

I save FIVE GVARS by doing that (RAET and flap rates), and even use another trick to allow the user to choose a switch or flightmodes for selecting the rate. The flightmodes are mapped on 10-90, and the curve will give a weight for each flightmode. But the values -100, 0 and 100 are reserved for a switch as input.

So the user (using my luascripts, but that is not obligatory) can set the source on the multiply line to the FM channel 10-90 values input, or a switch, and have all values stored.

lshems, 2021-07-03:
I think we must step back from the OPENTX design, and see where we can IMPROVE instead of just expand.

There are lots of hardly used values stored in the entire model. Let's see if we can be more creative in using those more effectively.

jfrickmann, 2021-07-03:
You always have to compromise, when you are cramming software into a device with only 128KB RAM. Having only 9 gvars is a huge compromise IMHO. And not having names for logical switches is a compromise. Another way to save memory could be to reduce the number of curves; yet another compromise.
Let's just change the number of gvars and see if there is a problem, before we start arguing over what compromises should be made to make room for them...

lshems, 2021-07-03:
How many values are reserved for all the curve points?  20x and 20 y values per curve, 16 of them? That's a lot of wasted space.

Can't we use curves more effectively? Can't we get a FLIGHTMODE source that maps from 10-90 so we can very easily use a curve as I described above?

Cant we include special multiplex function, saying the multiply should be applied only to the previous mixer line, and not to the result of all previous mixers?

Lots of possibilities

lshems, 2021-07-03:
And indeed, just increase. Nobody uses mixer scripts and thus should not use the lua compile option, so no issue at all probably.

lshems, 2021-07-03:
Mixer scripts need to store max 6 input source references, max 6 output sources, for 16 (or is it 32) lua mixer lines, so again 12*32 values....

pfeerick, 2021-07-03:
> Let's just change the number of gvars and see if there is a problem, before we start arguing over what compromises should be made to make room for them...

You silly you, you want actual data on if more GVs would be a problem, and if it's actually a compromise before you rabbit on about it being a compromise? :laughing:

jfrickmann, 2021-07-04:
Getting info on the increase in memory usage would be very easy - just fire it up and look at free memory under statistics. Then you can decide if it is necessary to cut somewhere else to make it fit. My guess is that it is negligible, and less than e.g. Mixer scripts, as l shems points out.
The one issue, though, is that it would affect the file format, and conversion functions would need to be made.

lshems, 2021-07-04:
As. I understood, the YAML format should prevent the need for conversion.

I've been pondering longtime to make a save and restore using a MUA script and the API. It would create tables for inputs, mixers, outputs, etc. On load they would be restored. Adding a few gvars would be completely transparent. Underlying technical format doesn't matter anymore.

It's the difference between an information model (what is functionally apppaering) and a datamodel (the underlying technical stored data.

RC-SOAR, 2021-07-05:
I agree, the restriction of 9 on GVARs is hugely frustrating, especially in the light of Ethos' unlimited VARs. Increasing to 16 would be a good starting point.

jfrickmann, 2021-07-05:
AFAIK, YAML is only implemented for X10 and X12, and the memory issue we have is for X7 and X9D. So for these, we would still need to change the storage file and make conversions.

_If_ we need to remove something, how about reducing the number of curves from 32 to 24?

@lshems, just to be clear - does your script rely on the name in Mixes, or is it the name in Outputs?

lshems, 2021-07-05:
Mixes. Output naming is essentially useless, as you they need to follow mixes anyway.

Why not name the outputs automatically after the mixers

lshems, 2021-07-05:
About YAML, can some clear this up??

It seems off topic, but is really relevant here in this context.

If there will be no general model storage format for ALL radios to exchange models in the field between radio types, a HUGE opportunity is missed.

In that case, I would really wonder what is stopping a Lua store/restore routine using the API.

Yeah, lots of settings not lua supported yet. But I don't see why energy should be spent on another storage format If it doesn't bring any advantages?

jfrickmann, 2021-11-28:
Now that we have YAML, I decided to try it out. And changing `MAX_GVARS` from 9 to 16 blew up a bunch of size checks and assertions datastructs.h and yaml_datastructs_funcs.h. I went ahead and changed the checks and assertions, and got libsimulator compiled. It seemed to run just fine, and there were actually 16 GVars! But the simulator cannot save YAML, and I have no way of telling if it would hose my model files and/or blow up the EEPROM if I compiled the firmware and loaded it on my radio.
@raphaelcoeffic @pfeerick - what do you say?

jfrickmann, 2021-11-28:
I have pushed the branch to #1189 so you can see what I am talking about!

ElectricRoxy, 2025-02-03:
Can we get even just 2 more gvars? If memory is an issue, there must be some fat to trim somewhere?

agnauck, 2025-02-19:
Would love to see more GVARS as well. There are many good use cases for them. In my case the reduce the number of mixers required which simplifies the model and reduces memory as well.