EdgeTX/edgetx #389: Cannot install latest Multi Module Firmware Update

MervDale, 2021-07-03:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

MervDale, 2021-07-03:
Using the latest nightly build for the TX16S on the Eachine radio unable to install the latest Multi Module 4 in 1 firmware.
Radio reports not a valid firmware file .
[image]

pfeerick, 2021-07-03:
I think you're trying to install it the wrong way, as I've never updated the MPM via the bootloader. In the past, I've used the SD browser when running the radio normally... It should give you an option to flash the internal module when you select the firmware file.

pfeerick, 2021-07-03:
I was able to update my the MPM on my Radiomaster TX16S just now, with a recent EdgeTX build. You're after this option, after located the file and selecting it on in the SD card browser... 
[image]

pfeerick, 2021-07-03:
Issue has been resolved (Discord)

closed 2021-07-03 by pfeerick