EdgeTX/edgetx #390: RCG Wishlist

pfeerick, 2021-07-03:
- more (at least 20) GV's (Overlaps with #388)

- another "layer" or "tab" similar to mixer lines, exclusively to calculate values without having to occupy mixer lines.
... e.g. progressive deflections or differentiations depending on flap positions.

- The possibility to use the results of the mixer channels or the above-mentioned "Calculation Lines" everywhere where numerical values can usually be entered (weight, diff ....).

- at several places longer identifiers as long as the display allows it
(leave it to the user if he wants to shorten it for reasons of backward compatibility)

- At least for mixer lines, the possibility to create a short help text per line under Companion; it is sufficient if these are stored in the model data and are only visible under Companion.

https://www.rcgroups.com/forums/showpost.php?p=47392211&postcount=111

lshems, 2021-07-04:
On the extra tab feature, may I comment that this is already 100% in line with the current LUAMIXERS.

Thanks

lshems, 2021-07-04:
As most things will ask storage, a rethinking of the storage concept is necessary.

Quick thought would be to DELETE all functional labelling information from the stored model, instead of adding. Then ADD the gvars.

But what with the labeling requirement then?

Well, I think it is perfectly possible to store them in (a) separate table(s), and combine them in the GUI with REDUCTION in eeprom size and INCREASE IN LENGTH of functional labels, WITHOUT a relevant impact on working Memory.

I use this approach on a complete new GUI written entirely in LUA, using the model API extensively.

This is proof this approach would work, since it doesn't matter if it is MUA or C doing that loading.

Only issue that needs consideration, is that I never use the model setup GUI at the same time as the Flight graphing GUI for instance. I solved that by making it part of one menu structure where you can't do both at the same time. Never been obstructed by that limit. On spektrum for instance the model is not even bound when doing some model adaptation.