EdgeTX/edgetx #392: Lua-Scripts in direct SYS access orderless

jogybaer0815, 2021-07-03:
TX16S Nightly d878efb

please output a alphanumerical order in the list of scripts (direct access via SYS)
Big Thanks

raphaelcoeffic, 2021-07-03:
> TX16S Nightly [d878efb](https://github.com/EdgeTX/edgetx/commit/d878efb376a64ea440e8b0c0e7c8cda1eb843d9f)
> 
> please output a alphanumerical order in the list of scripts (direct access via SYS)
> Big Thanks

I guess you mean the “TOOLS” page, right?

jogybaer0815, 2021-07-03:
Yes

pfeerick, 2021-07-03:
It's probably to do with the order they were added to the file system... they're in alphabetical order on my screen, but I've just copied them on after updating the MPM firmware... but yes, if it doesn't sort then that would be a nice touch indeed.

jogybaer0815, 2021-07-03:
Okay i am able to rearrange order by delete and recopy files (but have to do this every time a script was updated)

pfeerick, 2021-07-03:
Thanks for the confirm... so it is obviously just adding in file system order at the moment and needs 'sorting' before it's displayed.

PR #1150 (2021-12-05): sort lua scripts before building ui in radio tools
Fixes #392
closed 2021-12-05 by pfeerick