EdgeTX/edgetx #393: Logical switch V2 values can be out of range if V1 changes

pfeerick, 2021-07-03:
When doing a comparison like `a>x` with V1 set to GV9, and V2 set to -5, if you then change V1 to Batt, the transmitter incorrectly leaves the out of range value set (as GV has a range of -1024 to 1024, but Batt is 0-255, i.e. positive voltage) ... 

This appears to apply across the board to both colour and B&W radios... tested in RM TX16S and Jumper T-Lite. 

The suggested fix would be to put a range check in when v1 is changed, and reset the value to something within range if it is out of range.

lshems, 2021-07-03:
Can we or should we open an issue for checks when using the LUA API? Similar issues all over the place.

Ideally, if you start to introduce range checking, you should do that 'below' the API level.

Any interaction to change a value should call the same underlying API.

So a LUA call to change v1 should call the same range checking low-level API as the GUI call.

???

philmoz, 2026-02-09:
Not reproducible in 2.12 / 3.0 so I suspect it was fixed at some point.