EdgeTX/edgetx #395: Rename LUA build option to LUAMIXERS option

lshems, 2021-07-03:
Most people think they need to build including the LUA option.

Actually, the LUA build option enables the LUA mixerscripts, as well as the LUA script functions (is that so?).

Most people don't use them, don't need them.

Making the build option perhaps specific for lua mixers and lua functions, and naming them accordingly would avoid a lot of confusion?

lshems, 2021-07-03:
Perhaps include the luafinctions by default, depending memory impact, and leave the build option for the mixers.

I LOVE the potential of the lua mixers but until they are made fit for 'in flight' use, they are hardly used by anyone I assume.

pfeerick, 2021-07-03:
On OTX, I think it's main purpose was to enable the lua model scripts page... but it's certainly badly named because it isn't enabling LUA... 
> Model scripts A model script runs on a continuous basis once loaded into a model. As long as the model is selected, the script is running. You should never use a model script for calculating any sort of input that is vital to your model. Remember that scripts may be terminated at any time.

e.g. the betaflight lua script uses it to send RSSI and time in a background script...

lshems, 2021-07-03:
What are model scripts? Where are they stored on the Sdcard?

We are talking about the same thing: mixer and model scripts are the same?

lshems, 2021-07-03:
And using this option only to do that is very expensive in resources. You can do that in the telemetry script itself, so.... ?

Someone from betaflight here?

pfeerick, 2021-07-03:
Welp, it seems like it changed, they're using a special function now... or maybe it was something else... 

Anyway, you know as much as I do... I gave you the definition of a model script above :-P

lshems, 2021-07-03:
I will check if the function script is included or not in the gerenal build without Lua option. Reading the above I'm almost sure it is and the lua option is only enabling the mixer script (model script) page. They found out the resource impact and changed it to function script. :)

pfeerick, 2021-07-03:
Hm... telemetry and function scripts should work without any additional lua... it seems it's *only* needed to enable that model lua scripts page... which is then primarily for custom model or mixer scripts... 

http://rcdiy.ca/aa-mix-scripts-getting-started/

lshems, 2021-07-03:
Then let's call it LUAMIXERS build option instead of LUA.

I can't think of any useful application when it's not fit for flight, so this would allow most people to understand not to use it, freeing up memory and causing less crashes of lua scripts.

PR #1282 (2021-12-24): End the LUA != LUA confusion
Resolves #395

Summary of changes:
- Change `LUA=YES` to `LUA_MIXER=(Y/N)`
- Allow `LUA` to be come boolean, rather than tristate (Y/N/YES)
- Fix `LUA=N` compile failure
- Make CMake abort for anyone silly enough (moi!) to try `LUA=N LUA_MIXER=Y`
closed 2021-12-24 by raphaelcoeffic