EdgeTX/edgetx #397: Allow inverse calibration

lshems, 2021-07-03:
Lots of people complain about the direction of switches, sliders, pots, whatever.

Of course you can use a weight of -100 on an input.

But why not introduce a much simpler concept: three point calibration.

What does that mean? Well, users are first asked to CENTER any switch, stick, pot, slider.  ENTER to confirm. Then they are asked to put them into the MINIMUM position, so sticks down left, and all pots, sliders and switches in what the user wants as a Minimum value position. ENTER to confirm. Then the user is asked to put all in the MAXIMUM position.ENTER to confirm.

Make it an option to include switches and or trims in this procedure.

Result: 

> no complaints about inversed sliders, pots, ....
> No need for throttle reverse anymore
> One step further in eliminating legacy definitions like throttle, aileron, elevator and rudder. 
> 

Heck, you could even calibrate throttle from zero to 100 by setting the middle position with throttle stick down.

Lots of advantages.

?

lshems, 2021-07-03:
Throttle trim will then also become obsolete as option. I think alot of simplifications.

jmxp69, 2021-07-03:
Sims do this. Some correctly, some incorrectly, but that's the objective I think.

lshems, 2021-07-03:
So perhaps not such a bad idea?

jmxp69, 2021-07-03:
Not at all. Obviously the ability to reverse cannot be removed but if initial calibration can conform to the user…