EdgeTX/edgetx #398: Text boxes can't be tapped to edit

jmil-dev, 2021-07-03:
All controls (apart from textboxes) can be tapped twice to edit - first tap selects, second tap enables editing.

Text boxes can be tapped to select, but then the Enter button has to used to enable editing.

pfeerick, 2021-07-04:
It seems to vary by text box also, or model settings -> trainer channel edit boxes can be two tapped and edited... As can radio setup --> date fields... But not model setup -> model name, or model setup -> timer 1-> name

JimB40, 2021-11-11:
still valid?

pfeerick, 2021-11-12:
I don't think so... I suspect that it was resolved in 2.5 in through something to do with the on screen keyboard.

closed 2021-11-12 by pfeerick

JimB40, 2021-11-12:
but now single tap on text box opens keyboard. Can be closed.