EdgeTX/edgetx #399: Fast scrolling triggers haptic "min/max value" before it is reached

jmil-dev, 2021-07-03:
In a value field with a min/max scroll fast towards either the min/max. At a seemingly random point the haptic feedback for the min/max will start to trigger.

At this point any more scrolling increments/decrements the value and the haptic feedback keeps triggering.

I checked and it appears that the value is saved correctly so this is just a feedback issue.

Mantamatt1, 2021-07-03:
I can confirm same issue on tx16s rc3

pfeerick, 2021-07-03:
Yeah, it's as if the edge of the range has been hit... probably something to do with the scroll acceleration... i.e. sped up change would have hit the boundary, so it slowed down, but you still get the haptic/beep.

raphaelcoeffic, 2021-07-06:
> Yeah, it's as if the edge of the range has been hit... probably something to do with the scroll acceleration... i.e. sped up change would have hit the boundary, so it slowed down, but you still get the haptic/beep.

I think the correct behavior should be to clip the last increment, instead of rejecting it. And yes, the increment gets bigger the faster you scroll.

commit 8f55fb0f (2021-11-18): libopenui: NumberEdit set min/max if fast scroll goes out of bounds
Resolves #399
closed 2021-11-18 by pfeerick