EdgeTX/edgetx #402: after binding procedur binding sound remains

jogybaer0815, 2021-07-04:
TX16S nightly 4667b29 Internal RF
after activating Bind button and deselect Bind button binding sound remains

pfeerick, 2021-07-04:
When you say it wasn't recognised, do you mean with the internal RF set to "MULTI", that the status line said something along the lines of no multi telemetey? Do you need to reselect the protocol, or is setting the subprotocol sufficient? As I think there is an issue with the subprotocol list being "sticky" (i.e. you select the 7th subprotocol, for one protocol, but when you pick another protocol that only has 3  the wrong configuration is sent to the MPM and it gets confused. Just trying to work out if this is related or its own issue.

jogybaer0815, 2021-07-04:
Internal RF 
Mode                MULTI
RF Protocol       FrSky  D8
Module Status  No MULTI_TELEMETRY detected
...        
i reselected RF Protocol and now ModuleStatus changes to V1.3.2.86   
This Model was converted from OTX and i have not used this Model until today

And while I was making these entries, the touchscreen lost its touch capabilities and no longer dims after the set 5 seconds. If that's interesting - I know it doesn't really belong here ...

pfeerick, 2021-07-04:
Ah, maybe it's a conversion glitch then! 😱 

But at the same time... it still sounds somewhat related... When it happened, can you remember if it showed `Frsky` as the protocol, and `D8` as the subprotocol? Or was the sub-protocol blank?

Yes, the touchscreen going non-responsive is an issue we're trying to chase... we're pretty close as someone on the Discord has the hardware needed to test it, and it seems at present the touchscreen IC is locking up (which isn't a big deal of itself, as it *can* be reset, we're just trying to eliminate that its something the software is doing (or not doing) as the culprit before going that way.

jogybaer0815, 2021-07-05:
FrSky and D8 is showing. 
I had copied that model before checking this problem. I attach the bin-file

[models.txt](https://github.com/EdgeTX/edgetx/files/6761981/models.txt)
[model19.bin.zip](https://github.com/EdgeTX/edgetx/files/6762002/model19.bin.zip)

pfeerick, 2021-07-09:
Sorry for the delay. I just tried the file you provided, and was able to reproduce this perfectly... Even though the Mode, Protocol and Sub-protocol fields appear to be correct, `No MULTI_TELEMETRY` was being displayed. Re-selecting the sub-protocol and waiting for a second did not change that. Re-selecting the protocol triggered the module version and channel order to be shown. I'd have to iterate few a couple more times to see what else changes, but I noticed the RF fine tuning got reset from -20 to 0 when I re-selected the protocol. 

But at least this should be easily identifiable and fixable now! :)

pfeerick, 2021-07-09:
Can I just confirm that 'Low power mode' was supposed to be checked?

jogybaer0815, 2021-07-09:
Low power mode is/was not checked

pfeerick, 2021-07-09:
Ah ha... so that being turned on must have been part of the conversion going wrong! Thank you!

raphaelcoeffic, 2021-07-09:
> Ah ha... so that being turned on must have been part of the conversion going wrong! Thank you!

Should this be added to the unit tests regarding conversions?

pfeerick, 2021-10-03:
This was recently resolved

closed 2021-10-03 by pfeerick