EdgeTX/edgetx #404: The mixes screen doesn't have realtime feedback of changes

jmil-dev, 2021-07-04:
Similar to #307 the new UI doesn't have any way to see the values on the screen (either overview screen or mix detail).

pfeerick, 2021-07-13:
As a suggestion, when editing a mix, would pressing/long pressing MDL (and adding a button somewhere for touch) and bringing up the channel monitor be an appropriate resolution?

Or bar on the mix screen that shows the current mix output, similar to that shown for the channels?

Any other suggestions?

raphaelcoeffic, 2021-07-13:
> As a suggestion, when editing a mix, would pressing/long pressing MDL (and adding a button somewhere for touch) and bringing up the channel monitor be an appropriate resolution?
> 
> Or bar on the mix edit screen that shows the current mix output, similar to that shown for the channels?
> 
> But yes, i think we need to look at adding something similar to this to the mix screen proper...
> 
> [image]
> 
> Any other suggestions?

I would like this as a "bottom status bar". @JimB40 any comments on that?

JimB40, 2021-07-17:
Bottom Bar overlay is better.
1. Channel name is not displayed as we know it's selected one
2. Output bar id not displayed as we don't know if this mix is bound to any output
So basically just a bar. I would add % and us values n a way it's implemented in channel monitor
This mini-monitor can be visible in edit mode too so we can check changes without going back to mixer list

pfeerick, 2021-07-17:
1 & 2 and generall points all sound good.

3 - Huh? Since when can mixes not be bound to an output? (Inputs, sure, but mixes are CH specific). However, I still agree it's not needed - especially if it is made possible to access the channel monitor quickly again, via a second press of the MDL button (whilst in the model setup screen).

jmil-dev, 2021-07-17:
I like the bottom mix bar.

If space is a problem it could be at the left/right

Moksh-2000, 2021-09-24:
> I like the bottom mix bar.
> 
> If space is a problem it could be at the left/right

Yepppp i like too when you do something new or crazy add or multiply or something new then you should to know how's your pwm value react at specific point , i always use this for making something else @pfeerick hope you understand and @jmil-dev are right if problem with space then adjust somewhere but bottom is sure

pfeerick, 2022-09-09:
A Channel monitor button was added in 2.7 (now gone again :-O), and with 2.8 there is a toggle option at the bottom to show a mixer bar on demand for each mix.

closed 2022-09-09 by pfeerick