EdgeTX/edgetx #405: Problem in trainer page in Radio settings menu

BartJr26, 2021-07-04:
Hello everybody, 

I've a problem when i want to calibrate the neutral sticks position of the student radio in Radio Settings->Trainer page.

I can't press the word CAL to reset value of the sticks to zero what i was doing with OTX.
I can touch each value alone but as soon as I do it it turns green and the radio does not want to respond at all so i need to disconnect the battery to switch off the radio because i can't do anything else.

I've a Radiomaster tx16s and my firmware version is 2.4.0 (1dafc3aa).

Thank you everyone and sorry for my very bad english, i'm french and bad student in English class.

pfeerick, 2021-07-04:
Thank you for spotting that! It is implemented completely wrong - there is no button for calibrating and you should not be able to edit those fields - they should be for information only. 

The radio should not have locked up... when you selected those fields and they turned green for input, when you scroll the wheel it should have buzzed and beeped to indicate invalid input, and touching the top left corner or pressing the RTN button should have taken you out. If it did lock up, If it *did* lock up, that is a more serious problem.

pfeerick, 2021-07-30:
Video and another report of this issue: https://www.facebook.com/100004974581582/videos/361898888652348/

CamGenius, 2021-08-23:
I can confirm that it locks up for me as well on a TX16S Max

pfeerick, 2021-09-15:
Fixed by #716

closed 2021-09-15 by pfeerick