EdgeTX/edgetx #412: No mixer scripts

wimalopaan, 2021-07-06:
**Describe the bug**

no mixer scripts are available, the tab for the scripts does not show up in the icon bar

**To Reproduce**

cmake -DHELI=OFF -DTRANSLATIONS=DE -DPCB=X10 -DPCBREV=TX16S -DINTERNAL_MODULE_MLLTI=YES -DGVARS=YES -DLUA=YES -DALLOW_TRAINER_MULTI=ON -DCROSSFIRE=OFF -DXJT=ON  -DOVERRIDE_CHANNEL_FUNCTION=OFF ..

lshems, 2021-07-06:
You need to compile with Lua option. I will check if they work for me this evening

lshems, 2021-07-06:
Is there a typo on the MLLTI part. That might ruin things

wimalopaan, 2021-07-06:
> Is there a typo on the MLLTI part. That might ruin things

its not, was wrong copy-paste

wimalopaan, 2021-07-06:
it is simply not implemented

lshems, 2021-07-06:
Fun. Only for colour or also missing for B/W.

lshems, 2021-07-06:
Someone must have decided to take it out. Either before or after the fork, that's the question. Lol

raphaelcoeffic, 2021-07-06:
> Someone must have decided to take it out. Either before or after the fork, that's the question. Lol

Really nothing intentional: it has just been forgotten, and that was before forking.

lshems, 2021-07-06:
I forget my own birthdays, so no big deal. I do wonder if it can be brought back in a more reliable way. Lol

wimalopaan, 2021-07-09:
> > Someone must have decided to take it out. Either before or after the fork, that's the question. Lol
> 
> Really nothing intentional: it has just been forgotten, and that was before forking.

I hope you can get it in very soon ;-) 
This is basic functionality and should be supported by etx in 2.4.0.

sfjuocekr, 2021-07-14:
There were a lot of things on the 2.3 master branch that had not trickled down into the 2.4 branch yet before it got forked!

PR #436 (2021-07-14): [color] mixer scripts screen
Todo:
- [x] display only one of .lua/.luac in filechoice
- [x] display proper script status on list page

Resolves #412.
closed 2021-07-14 by raphaelcoeffic