EdgeTX/edgetx #417: Warnings can be dismissed by touching anywhere on screen

jmil-dev, 2021-07-07:
If you start the radio and a warning appears (throttle/switch/failsafe) it can be dismissed by tapping the screen anywhere.

This seems a little unsafe as the screens are capacitive and respond to a very light touch. 

I think there should be a specific onscreen UI element that needs to be tapped - possibly with "an are you sure" confirmation, especially around throttle.

jmil-dev, 2021-07-07:
Thinking about this a bit more - slide to dismiss?

raphaelcoeffic, 2021-07-07:
> Thinking about this a bit more - slide to dismiss?

In the sense of making it harder / be sure the user read what is on the screen?

jmil-dev, 2021-07-07:
At least for the throttle make sure it is deliberate.

jogybaer0815, 2021-07-07:
ooops - if i change the model there isn't a check if switches are in risky positions!
That is a big risk.
If switch warnings are set, only correct switch-positions should activate the model

Mantamatt1, 2021-07-07:
Esc's dont arm unless they are on 0 throttle. My only input would be to have a specific place saying OK or DISMISS rather than the whole screen. Other than that I think it's fine. Having warnings for warning just annoys me!

pfeerick, 2021-07-07:
> Esc's dont arm unless they are on 0 throttle. 

Not always so. I have some ESCs that do not have the arming check, and just do what they're told. Also, ESCs are not that only thing that can be on the throttle channel... I.e. glow motor throttle servo. You also have to take into account switching to the model *after* it's powered on (and the failsafe arms the ESCs). btw, if you don't need the warnings, you can turn them off. 

The "making it harder" isn't about if the user read the warning - it's about ensuring the page was deliberately dismissed - hence why swiping or pressing a specific control on the screen would be a good idea.

Mantamatt1, 2021-07-07:
I have many many models, planes helicopters quads ect. And I'm a member of
2 clubs so may models, not one esc will arm with throttle applied.
please can you tell me what brand of esc you have that can do that as I
have a project that would be useful to have a constant rpm without having
to reset?

And yes I agree,  have a specific place where you tap to acknowledge
unexpected switches or stick positions rather than the whole screen.
All I was saying was having a message after, thats warning you, that your
ignoring the warning, would be anoying!

Lol that was hard to word!

pfeerick, 2021-07-07:
lol... indeed... rest assured,,, we don't need to warn that you didn't read the warning!

I'll see if I can find them... if I remember correctly they were just generics and didn't even have any logo or branding heatshrink... so unless I can figure out where and when I bought them (was quite a few years ago, and hopefully I didn't throw them out as they are bloddy dangerous)... the description would probably be 30A yellow heatshink ESC so no very helpful! 😮 

The ability to program your own ESCs what why I liked the SimonK firmware... if you could get some compatible hardware you could reprogram it behave however you want/need.

Mantamatt1, 2021-07-08:
I've only played with putting blheli on generic escs not the Simon k. And
blheli does not have that option. I have a water pump that is driven by a
brushless motor and I have a servo tester to operate it. When I turn it on
I have to zero the servo tester to arm then turn it up to run the pump.
Would be good if I can just turn it on and go.

pfeerick, 2021-07-08:
Yeah, BLHeli only gives you the configuration options they provide. SimonK can be self-compiled (https://github.com/sim-/tgy) so if you can figure out how to drive it,  you can literally make it do *whatever* you want. 

Alternative would be an Arduino Nano with a potentiometer, programmed to act like the servo tester and have a varying PWM output with respect to the pot position, but when it initially powers on emitting a 1000ms (or thereabouts as needed by the particular ESC calibration) PWM for a short period so that it arms.

jmxp69, 2021-07-21:
Throttle lock logic works regardless of warning screen feature. How many safeties do we need?
1) You should have an effective throttle lock--ok
2) You should have switch monitor to warn you that your throttle is up (and your throttle lock may not be engaged)--ok
3) You should have to dismiss this warning upon boot--ok
4) You should have a special place on the screen to dismiss this warning upon boot--Where does it stop?

JimB40, 2021-07-21:
Hey we are Edge :)
Modern drones firmware will not allow arming with throttle applied.
I have throttle & switches warn off in my quads.
So throttle & switch warning may appear if option is selected in model setup but surely no warning that any of these options was set to off

raphaelcoeffic, 2021-07-21:
> but surely no warning that any of these options was set to off

A revival of "Warning: you disabled the telemetry warnings!"? That would be fun, indeed.

raphaelcoeffic, 2021-07-21:
It seems the consensus is: "enough is enough", we have a sufficient level of checks and warnings, right?

jmil-dev, 2021-07-21:
I think my bug is being sidetracked :) I'm not advocating for more warnings, I'm suggesting that warnings should not be dismissed by touching anywhere on the screen, from memory, not all buttons can dismiss all warnings on OpenTX either.

If you add this bug to #420 you can see a user not even spotting a warning message - tapping the screen will remove the warning at the same time as it turns it on.

jmxp69, 2021-07-21:
> At least for the throttle make sure it is deliberate.
> 
> As it stands right now you can turn the radio on with throttle up, brush the screen and have a spinning prop.

My perspective doesn't change. At some point the user has to be accountable. The device needs to conform to the user, not the other way around. Any single button on OTX can dismiss a warning screen.

1) PILOT: Physically inspect throttle stick, throttle lock mechanism before power-up, and other switch positions worth warning.
2) Use good logic 
3) Use switch warnings if you chose
4) View boot-up warning screen with audible call-out
5) Press button or screen to bypass

There are literally _five_ levels of safety to prevent inadvertent actions/prop-spinning. Six levels if you employ checklists. IMHO it simply does not need to be more difficult to operate the radio.

jmil-dev, 2021-07-21:
That's why you can choose to disable the warnings...

But if you want them on they should actually work!

jmxp69, 2021-07-21:
They do work. You're blaming user error on the radio. Bad form.

pfeerick, 2021-07-24:
> Any single button on OTX can dismiss a warning screen.

Actually, no. i.e. PAGE, TELE, SYS, MDL, NEXT, PREV don't skip warning screens. only RTN and ENT, and I think some warnings can only be skipped with RTN. In this respect, the prompt to 'press any key to continue/skip' is actually misleading.

I think as far as mixing #420 into this equation, that is beside the point - as it is a very specific case, and is also a user caused. There is only so much hand-holding you can do without expecting _some_ user accountability.

I think we need to simply come to some consensus as to whether:
1. The arrow on the screen indicates a touch area. If so, you should be required to touch that specific area of the screen to dismiss the warning, whatever it may be. It will be annoying at first, but you'll train your finger to hit the hitbox quickly enough if that is the way things go. 
2. That touching the whole screen is acceptable for dismissing a warning (perhaps for more than x time period - i.e. 200ms or something to mitigate 'brushing the screen' dismissals?) 

I for one for the most part don't care either way - my quads won't let me arm with the throttle high either because of arming switch or needing the throttle low as part of the arming sequence. My helis won't bite me either as they require throttle low. My planes won't either as they'll at me profusely until the throttle is low. So even if I did something stupid, my gear will protect me from myself. But that is just *my* use case. And I won't even get into the "Transmitter on first, plane second... so why will the plane burst into life?" argument.

jmxp69, 2021-07-24:
Actually yes, I just confirmed it and on my radio I can dismiss switch warnings with any button including telemetry.

pfeerick, 2021-07-24:
I should have been morè specific, as I only quickly checked the throttle warning, and with the TX16s, only the ENT and RTN buttons will dismiss it. Perhaps the switch warnings is not so fussy, which IMO is wrong, as it means inconsistent UI behaviour.

jmxp69, 2021-07-25:
Throttle and warnings can be dismissed with any button on OTX:
https://youtu.be/7_v92YE5egM

While I didn't film it, Throttle can also be dismissed with the jog dial, model button, and system button.

Eldenroot, 2022-10-09:
This should be possible only with ENTER or RETURN buttons... not?

pfeerick, 2022-10-09:
No. If touch is available, it is an option also. The issue here is that it is the whole screen, hence:
> I think there should be a specific onscreen UI element that needs to be tapped

Which used to be the case, at least visually. 

[image]

philmoz, 2023-01-19:
In the current branch the SYS, TELE, and MDL buttons navigate to their respective UI pages even if a switch or throttle warning is being displayed. The PAGE up and down function can be then used to navigate within the other pages if they are opened.

These buttons do not dismiss the warnings so the 'Press any key to skip' message is incorrect.

IMHO the user should not be able to do anything else until all warnings are dismissed, and any button press should dismiss the warning (as it says on the tin).

PR #3061 (2023-01-25): fix(ui) - Button and touch handling for throttle / switch alert screen.
Fixes #417

For throttle & switch alerts:
- Use lvgl objects for text strings.
- All buttons work to close the alert, buttons will not open other UI pages over the alert.
- Touch screen can close the alert if touched for 200ms for longer, brief touch will not close the alert.
- Display the 'press any key...' message on the switch alert page.
closed 2023-01-25 by pfeerick