EdgeTX/edgetx #418: Backlight behaviour is unexpected if using a global function

jmil-dev, 2021-07-08:
Set the backlight to "Both", assign a global function to set the backlight brightness, assign it to a control (I used S2). Turn the backlight down so the screen goes off. Leave the controls, then wait, the backlight turns back on, and it goes off if you touch a control.

Not really a high priority though and based on Discord is also a bug in OpenTX

jogybaer0815, 2021-07-08:
yes, but that is correct, because after the set time the value for OFF brightness is switched. after actuation, the screen is switched back to ON brightness.
So if the OFF is higher than the ON, you have the effect described

pfeerick, 2021-07-08:
Not quite... what is happening is if in the given scenario, when you set S2 to 0%, with ON brightness say 75%, and OFF brightness 25%, the screen will turn back on to the OFF brightness after the delay has elapsed. 

In all fairness, it's not really a bug, but an expected behaviour. If you want to use the knob to set the brigness with the settings structured as they currently are, the mode should actually be set to 'ON', since you want the knob to be setting the brightness, and for it to not change. 

What I think you're actually after, is for the radio to support setting the brightness via a pot natively, and changing the ON brighness, so that you have the ability to have a timed dim *AND* controlling the brightness. Does that sound about right?

If so, I would suggest that there be a dropdown to allow linking the on brightness to a GV or available POT/SLIDERs, which can control the on brightness. 

Just as there needs to be a mechanism to link the volume right to a POT right there in the audio settings, rather than as a special function.

pfeerick, 2021-07-08:
For reference: https://github.com/opentx/opentx/pull/7680

gagarinlg, 2021-11-07:
> Actually thinking this through a bit further, you could say there is a bug, in as far as letting the brightness go down further than the "OFF brightness". ?

With my last commit to #998 it is not possible anymore to set the brightness below the dimmed brightness, except when the display mode is set to ON. Thich should resolve the "bug" part of this issue.
I am not sure, if it is completely resloved, though.

philmoz, 2023-12-13:
PR #4060 scales the backlight value from the pot/slider to fit between the OFF & ON backlight settings.

Does that resolve this issue?

pfeerick, 2023-12-13:
At first thought, yes... I'll check properly later.

commit db410eb2 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit 8798b51e (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit ba0fb945 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit ab3cb457 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
closed 2024-01-11 by github-actions[bot]