EdgeTX/edgetx #420: Touch is accepted with the backlight off

jmil-dev, 2021-07-08:
If the backlight is off a touch will both turn the backlight on and be registered.

Ideally a touch is required to turn the backlight on and then a second touch is needed to interact with the UI

gagarinlg, 2021-11-05:
My backlight solution can be found here:
[https://github.com/gagarinlg/edgetx]
in the main branch.
It does work on TX16s and should work on NV14, at least it compiles and it is not to many changes.
Compilation for non color lcd/non touch radios is also OK.
If this is an acceptable solution, I would create a pull request from it.

MervDale, 2021-11-06:
Please don't take this as criticism but it seems to me that there would be more pressing issues that would benefit from programmers time.

To me that this does not seem to be a major issue in that I would say that the main concern of having the back light off is to conserve battery power when flying and reactivating it without affecting the touch screen. 

This can all be done with the current OTX options as I see it. Please correct me f I am wrong.

I have mine set to 50 sec to off and back on to half power "approx" with keys so just a roll of the wheel on the TX16S brings it back on with the option of turning it up or down with a GF on S2.

Not being a programmer could you explain how you option works please.

gagarinlg, 2021-11-06:
Feel free to say what you think. I very mich prefer an open discussion. As for the reasoning:
1. it is a hobby, not a job, so I decide what I do with my time 😜
2. I just started participating in this project, so I selected an issue with little to no impact on the real rc stuff, to get to know the software internals. 
I do dim my backlight faster, but that is a personal preference. 
The main concern is, that e. g. on the NV14, there is no way to activate the backlight without creating an input event, as there are no buttons.
When you are in certain lighting conditions it can be very hard to discern what is on your screen and then it is helpful to not create an input event.

MervDale, 2021-11-06:
By all means go for it, I have no issues with your option. I was just thinking out aloud,  it seems Australian English doesn't  convey the same thoughts as other versions, my bad.

gagarinlg, 2021-11-06:
😂 There are a few distinct differences between german and australian English... 
As I said, I like an open discussion. So no offence taken.

raphaelcoeffic, 2021-11-06:
> Please don't take this as criticism but it seems to me that there would be more pressing issues that would benefit from programmers time.
>
> To me that this does not seem to be a major issue in that I [...]

What do you think of what is proposed here: https://github.com/EdgeTX/edgetx/issues/420#issuecomment-954643823 ?

gagarinlg, 2021-11-06:
> What do you think of what is proposed here: #420 (comment) ?
If I am right, this is essentially, what is there now, except for being able to set the backlight timeout to 0, which, in combination with setting the dimmed brightness to 0 leads to an unusable state.
So I will change the minimum backlight timeout to 5.
Also I will try to disable unused control elements, so when backlight is set to be always on, everxthing but the on brightness control element will be disabled.

PR #1000 (2021-11-16): Touch backlight handling
this PR needs https://github.com/EdgeTX/libopenui/pull/34 to compile and work correctly.
With this PR touch handling is modified to prevent touch events from happening, if the backlight is off.
For this to work, touch drivers now need to return the TouchEvent structure instead of writing into a libopenui global variable.
The interface between touch driver and backlight control now hast to go through libopenui, to be able to filter touch event there, without activating the backlight before libopenui can handle the event.

The touch driver for NV14 got the necessary modi [... cut]
closed 2021-11-16 by pfeerick