EdgeTX/edgetx #421: Internal GPS not available

stefaanbolle, 2021-07-08:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
I modified with an Internal GPS (BN-220) on TX16S AUX2. In openTC 2.3.13 this works and displays correct values with the 3djc GPS test script.
With EdgeTX the 3djc GPS test script, the widget remains blank. Or there is a problem with the script in EdgeTX or with internal GPS (did not find a possibility to enable this in the firmware)

**To Reproduce**
Steps to reproduce the behavior:
create a widget using the 3djc GPS test script

**Expected behavior**
A clear and concise description of what you expected to happen.
The 3djc GPS test script should reveal internal GPS info

** Info **
 - EdgeTX version (e.g. 2.4.0 nightly 7f472ff)
 - RM TX16S

pfeerick, 2021-07-08:
You probably need a build that supports the internal GPS, as you would have had to tick the "Internal GPS" option in the OpenTX companion in order to get a GPS supported build of the OpenTX firmware.

rotorman, 2021-07-08:
To see if your firmware has internalgps option baked in or not, you can also check under SYSTEM -> VERSION.
The OPTS line should list `internalgps`.

pfeerick, 2021-07-08:
This is a build of [firmware_7f472ff_TX16S_internalgps.zip](https://github.com/EdgeTX/edgetx/files/6784168/firmware_7f472ff_TX16S_internalgps.zip) #7f472ff with `INTERNAL_GPS=YES` as the nightlies don't have it enabled ... do you want to give that a try and see if it works... for you?

stefaanbolle, 2021-07-08:
Thanks! will try the pfeerick build and install build environment in a VM

closed 2021-07-08 by stefaanbolle

stefaanbolle, 2021-07-08:
Works like a charm now, thanks!

pfeerick, 2021-07-08:
Fantastic! If you setup up a docker build environment, you'll find it pretty easy to build your own firmware on demand, if you need specific options (like `internal-gps`, or tweak some of the other settings).