EdgeTX/edgetx #429: Multi DSM config options strange/incorrect

olliw42, 2021-07-10:
On the T16/Tx16s transmitters with an internal multi module, the configuration options for the DSM mode are **_NOT_** equal for OpenTx and EdgeTx.

OpenTx:
[image]

EdgeTx:
[image]

I find this most discomforting and also highly confusing ... I mean, I do not really understand all the options for OpenTx nor for EdgeTx, but now they are even different besides one and the same module and one and the same MPM firmware is used ... I can't make any sense of it.

Hence, question:

Is this difference intentional, i.e., has a good reason? Or is it actually not intentional?

If it is not intentional, and it should look the same as for OpenTx, then I would make a PR which streamlines this (or if it should be as in EdgeTx I would make a PR for OpenTx LOL). However, since I understand this MPM stuff only poorly, I thought I should better ask beforehand.

So, what's up here? Yes or No?

pfeerick, 2021-07-10:
As far as the ordering, probably no reason for them to be different, would just be the order they were re-implemented. 

There are some options commented out in the code, so I think there is more stuff needing to be implemented/added and that max throw option should probably be a checkbox.

pfeerick, 2021-07-10:
I would work backwards from this, and verify would should be present for DSM ...  https://www.multi-module.org/using-the-module/protocol-details/dsm

olliw42, 2021-07-10:
I'm not concerned about the reordering, I can intellectually handle this LOL ... I'm concerned with that there are different options (and also that max thrw can be a checkbox or a number ...)

concerning the linked page, it seems to me that this is what one finds in OpenTx ... but ... what is the actual true and correct status ?

pfeerick, 2021-07-10:
That is the MPM documentation, *not* the OpenTx documentarion. Yes, it shows how things are interfaced via opentx, it it still tells you what options are supported by the protocol. 

Otherwise, you will have to go further down, to the actual module spec, and try and match things there. Not so easy if you don't understand MPM stuff to begin with, but will certainly be a good learning exercise about it (re: DSM options) => https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#dsm---6

olliw42, 2021-07-10:
so, to get a simple yes/no ... you are saying, yes, the EdgeTx list of DSM options do not look correct?

pfeerick, 2021-07-10:
Lol, yes, they are not correct. As mentioned before, the max throw for starters should not be a checkbox. Also, the auto option is missing from the subprotocol dropdown.

olliw42, 2021-07-10:
>  yes, they are not correct. 

ok

so I know that I can work on them and make the aligned with what one expects to see, so I will do ... :)

olliw42, 2021-07-13:
@pfeerick 
pl un-assign me from this task
I looked into the MPM documentation for DSM, and, with all due respect, this is just perfect glibberish
I really do not have the time to spend the ages to decipher this

all I possibly could do is to make EdgeTx to do exactly what OpenTx is doing for DSM, this would be easy
but if the expectation is - which I somehow got from the above - that EdgeTx should do what the MPM documentation says should be done, then I definitely need to drop out

I have no idea that the goal here would be, just to mimic OpenTx or follow MPM docu, so I'm on stall.

raphaelcoeffic, 2021-07-13:
> @pfeerick
> pl un-assign me from this task
> I looked into the MPM documentation for DSM, and, with all due respect, this is just perfect glibberish
> I really do not have the time to spend the ages to decipher this
> 
> all I possibly could do is to make EdgeTx to do exactly what OpenTx is doing for DSM, this would be easy
> but if the expectation is - which I somehow got from the above - that EdgeTx should do what the MPM documentation says should be done, then I definitely need to drop out
> 
> I have no idea what the goal here would be, just to mimic OpenTx or follow MPM docu, so I'm on stall.

So basically we have 2 options: copy the current behavior from the OpenTx code, or ask Pascal for explanations.

pfeerick, 2021-07-13:
In the first instance, I would have expected to adopt the current behaviour from OpenTX as much as makes sense given different interface options. Beyond that, anything does doesn't make sense would require questioning Pascal as Raphael suggested, or alternately MRC3742 as has also been responsible for some MPM updates to OpenTX.  Obviously entirely up to you if you want to work on this or not, and it's not like it was expect to be to be done tomorrow or anything like that ;)

olliw42, 2021-07-13:
> entirely up to you if you want to work on this or not

I would work on mimicking latest OpenTx, and would do it soon, but I would not work on ensuring it's according to latest MPM, not now and not later

well, so a clear yes or no, pl

if I would make EdgeTx to (only) mimic latest OpenTx 1-to-1, would this be considered for merge now (= soon LOL), yes or no

I clearly want to avoid doing something which is then just sitting there for long :)

raphaelcoeffic, 2021-07-13:
> > entirely up to you if you want to work on this or not
> 
> I would work on mimicking latest OpenTx, and would do it soon, but I would not work on ensuring it's according to latest MPM, not now and not later
> 
> well, so a clear yes or no, pl

Yes!

> 
> if I would make EdgeTx to (only) mimic latest OpenTx 1-to-1, would this be considered for merge now (= soon LOL), yes or no

Yes!

> 
> I clearly want to avoid doing something which is then just sitting there for long :)

We are definitely trying to avoid this, regardless of which PR this is about. Fortunately, the touch sensor nightmare is coming to a temporary end, so that we should be able to move on and be faster about new PRs.

olliw42, 2021-07-13:
> > if I would make EdgeTx to (only) mimic latest OpenTx 1-to-1, would this be considered for merge now (= soon LOL), yes or no
> 
> Yes!

this was clear enough for even me understanding it :D

raphaelcoeffic, 2021-07-13:
> Receiver number lets you go past 22

max channels depending on module / protocol is a pretty mess right now. I gave up understanding the code last time. This needs a major rework (3-4 different and redundant methods / tables are used….).

pfeerick, 2021-07-13:
Yeah... I don't know how that works at the moment, but it _is_ working... as it caps at 12 for DSM. It's the receiver number that needs capping, but it isn't on OTX either... and I would be willing to bet isn't quite configured _right_ on OTX anyway, as it also lets you go past 22, and then there is this whole "set to zero if you want to ignore the receiver ID", and the receiver id being global mess. 

But its certainly one problem to avoid for now... I don't know what happens with the MPM though if you try to bind on say model 25 though... it doesn't seem to complain that the module status is invalid or anything, so maybe keeps the number in range somewhere else. 

It really only comes to four things needing to be added / fixed AFAIK:
* servo refresh rate checkbox missing, 
* max throw should be checkbox
* subprotocol is missing options 4 and 5 (AUTO and DSMR_1F). 
* Probably remove the autodetect format checkbox as it's not present in OTX any longer and redundant since it's the AUTO subprotocol.

pfeerick, 2021-07-18:
@olliw42 I'm afraid we couldn't wait for this as we want to push 2.4.0 out with this. If you get a chance, can you see if #471 resolves this for you?

closed 2021-07-19 by pfeerick