EdgeTX/edgetx #430: Output Widget: Displayed labels do not make sense/are incorrect

olliw42, 2021-07-10:
In the output widget, the displayed labels for each channel are highly confusing to a point that I would call them incorrect.

I do have a very simple inputs/mixes/outputs configuration:

[image]
[image]
[image]

And this is what I see in the outputs widget

[image]

Obviously, the displayed labels should mirror the outputs, while they mirror the inputs. Accordingly, I get the strange behavior that then I move, e.g., the throttle stick the "Ele" bar is moving ... 

What I would have expected to see is the very same labeling as in the outputs tab, i.e., Ele, Thr, Ail, Rud, ch5, ch6, ch7, 08, 09, and so on.
In OTX this is in fact so in the sense that it displays CH1,CH2,CH3, and so on.

I looked at the code but don't understand the whole indexing well enough to make the correction myself. The lines of issue are here
https://github.com/EdgeTX/edgetx/blob/2.4/radio/src/gui/colorlcd/widgets/outputs.cpp#L73-L79

closed 2021-07-15 by pfeerick

olliw42, 2021-07-16:
@pfeerick 

many thx for the changes!

it is showing now CH1,Ch2,... as it does with OTX, which is clearly MUCH better than before ... 

BUT: why not showing the name of what CH1,CH2,... are ??? I think this was the intention in the first place to bring in such an improvement

I hence think the issue could/should be reopened (or another issue opened?)...

pfeerick, 2021-07-16:
No, that is not the case. If the name of the output is not named it will not show the name, and will show the channel. If you name the output - i.e. go to the outputs page, and name the outputs, it will show the name the output. As was shown in the after image of the effects of the PR, the channels 1 and 4, which were named "O T" (O for output, space, T for Throttle), and "O R", it will show the names for them, and in fact would do so all along. It was a downright mistake before that it showed the input names. I don't think it will be practical to use the name from a mix, as you can have multiple rows for a single mix, so which one do you extract the name from? And to be frank, mixes also `=!` outputs. Maybe we need a mixes widget as well... perhaps a LUA one? ;)

[image]

If there is still a valid option, by all means, new issue / PR ;)

olliw42, 2021-07-16:
hmhmhm... I see... I indeed confused mixes with outputs

the naming things doesn't make much sense to me, e.g. there are Ail,Ele,Thr,Rud names for source and inputs, which are given to the first four entries by default, but not so for mixes, which are named CH,...,CH4. If the naming logic would be consistent the first four should also be named Ail,Ele,Thr,Rud by default. Anyway, but I agree this is general issue beyond the present state.

thx for clarifying this, my fault
sorry for having stolen time from you :)

pfeerick, 2021-07-16:
Btw, be aware of this... :-O https://github.com/EdgeTX/edgetx/issues/460

olliw42, 2021-07-16:
> Btw, be aware of this... :-O #460

yeah ... noted it too

back to the before, if I may make a suggestion, I guess I would NOT display the 01,02, etc if there is a name to the output ... it's really superfluous if the user has decided on a name...

if this would be agreeable, I would be happy to raise a corresponding PR ...

pfeerick, 2021-07-16:
I think the naming only makes sense when you starting using more convoluted programming in OTX/ETX... Then it becomes a requirement ;)

i.e. Input names are used for inputs in mixes. Mixes are output to CHs, so they are listed by the row of the channel they belong to, but the various mixes can get named so they make sense also. 

I.e. at a glance, I know what inputs are being used on CH3 and CH4, and I can see that CH5 is a multi-line mix, what each mix line is for as I named each mix, and in this case, the actual value is controlled by a flight mode. 

I don't think removing the channel number is a good idea, as you can configure the widget to show channels from a given starting channel, so this ensures you remember this has been done. If you could make it so it were an option, which I think is quite doable, I would love to see a PR improving the widget further ;)

[image]

olliw42, 2021-07-16:
I think I understood this now
what I meant is that CH1,CH2,CH3,CH4 should also be named Ele,Ail,Thr,Rud by default exactly as it is done for sources and inputs ... the difference between source and input is made by the icon in front of the name, and so it could be also for the mixes, i.e. the first four could have the mixes icon + Ele,Ail,Thr,Rud name ... this would make it consistent across the levels...

pfeerick, 2021-07-16:
If you mean the outputs should be also named by default, then yes, I think that should happen. I am surprised outputs weren't named by default also, given that inputs and mixes are. 

Other than that, I don't think anything else should be changed at this point, unless it is done through options.

olliw42, 2021-07-16:
> If you mean the outputs should be also named by default, then yes, I think that should happen. I am surprised outputs weren't named by default also, given that inputs and mixes are.

yes, this was what I was trying to say :)

> Other than that, I don't think anything else should be changed at this point, unless it is done through options.

clear

olliw42, 2021-07-25:
done

PR #455 (2021-07-15): Make outputs widget use outputs names
Resolves #430.

Basically, current behaviour would incorrectly use the names of INPUTS if a OUTPUT was not named. 

i.e. Given that the first and fourth outputs have been named (CH1 = "O T", CH4 = "O R"), you can see that two and three are using the input name (I2 = "I A", I3 = "I E"). 
[image]

This is ok, except for those times when your inputs don't match the order of your mixes, and hence output names or numbers. 

This makes it so they instead fall back to showing the channel number, consistent with the 'unnamed' form of outputs.
[image]

Thus also co [... cut]