EdgeTX/edgetx #431: Model Setup, Model Name: Extend available chars to what one has in OpenTx

olliw42, 2021-07-10:
In the Model Setup, Model Name text entry field, one currently can enter only the special characters `:;<=>` (besides the expected chars ' ','A'-'Z','a'-'z','0'-'9').

but: who the f... wants these chars in a model name !?!?!?!

in OpenTx one in addition can use the characters `_-.,`, and I for instance miss the '_' and '-' much

I thus raised a PR to lipopenui, which allows to (1) specify an arbitrary set of extra characters and (2) allows to modify this set as needed by allowing to supply the desired set as a string to the constructor. The chosen approach ensures compatibility with current behavior, but also allows most flexible adaption, I believe.

This PR to libopenui needs to be complemented with some few changes to the EdgeTx code, which however meaningfully can only be PR-ed when the changes to libopenui are merged (since it needs updating the submodule). I thus will raise that complementing PR when this should happen.

The result of both PRs will be that in the model name one now can use the extra characters `_-.,:;<=>`, but this would be most easily modified to any set the UI experts think it should be.

raphaelcoeffic, 2021-07-12:
@olliw42 the first PR is merged, please go ahead with the second! But we need to be fast on that one, as release is imminent ;-)

olliw42, 2021-07-12:
I'll do now, give me few minutes... ;)

olliw42, 2021-07-12:
done, so I close

closed 2021-07-12 by olliw42