EdgeTX/edgetx #432: Display the name of a mixer-script when selecting as source

wimalopaan, 2021-07-11:
When selecting a mixer script as source of e.g. a mixer, only the internal (coded) name of the script displayed. That is confusing if one uses more than one instance of the same script.
One can give each instance of a mixer script a name, but this name is not displayed when selecting that script as source.

I would suggest to use the coded name, if no name has been configured, but displaying the configured name if there is one instead of the coded name.

raphaelcoeffic, 2021-07-12:
I think I need some context, which radio is this? On the color radio mixer scripts are yet back.

wimalopaan, 2021-07-12:
Yes, this is for colorLCD. 
I know that mixer scripts are not yet back (yes, please make this possible as soon as possible!), so this is more a reminder to change the behaviour inherited from OpenTx.

I don't have access to monoLCD, but I think the issue is actually present there.

raphaelcoeffic, 2021-07-12:
Do you mean actually the outputs the script? So basically you want to scope the output name?
So basically `Script1/Value1` vs `Value1`?

wimalopaan, 2021-07-12:
Say <output> is the name of the output-table (which is the same for all instances of this script) and <name> is the name one can configure in the script-dialog, the displayed name should be ideally <name>/<output>, but I think this would be too long in most cases.

So I think it would be sufficient to display only <name> iff <name>!=nil, else <output> (as it is now).

wimalopaan, 2021-07-13:
In the mixer definition the name is missing
[image]

Here "SchottDir" is the name in the output table. The configured name is missing

[image]

Here it is as intended:

[image]

raphaelcoeffic, 2021-07-13:
This is not the name of the table, but of a single output. But you can have multiple outputs per scripts, did you know?

pfeerick, 2021-07-13:
What's the name for... is that like the name of that instance of the script? If so, what happens if you call the same script in LUA1 - LUA4 with a different name for each? How would you identify the output from each? I've not used custom lua/mixer scripts before, so I have no idea how they are usually used

wimalopaan, 2021-07-13:
> This is not the name of the table, but of a single output. But you can have multiple outputs per scripts, did you know?

Oh, thats cool. No, I wasn't aware of that!

raphaelcoeffic, 2021-07-13:
> > What's the name for... is that like the name of that instance of the script? If so, what happens if you call the same script in LUA1 - LUA4 with a different name for each? How would you identify the output from each? I've not used custom lua/mixer scripts before, so I have no idea how they are usually used
> 
> Thats why I suggest using the instance name together with the output name.

That is understood. So basically “InstanceName/Output”.

wimalopaan, 2021-07-13:
> That is understood. So basically “InstanceName/Output”

wimalopaan, 2021-07-13:
Sorry, was an accidential close ...

pfeerick, 2021-07-13:
If I had a dollar for every time that happened 😆 I think we've been trained by Windows and Linux UIs (Mac seems to be more in line with Github - colourised and to the right side) that the 'ok' button is on the left side, so hit close with comment by mistake so easily!

wimalopaan, 2021-08-28:
Is there something going on here?

pfeerick, 2021-09-02:
I'm going to tentatively put this down for 2.6. Mixer scripts need a kick up the backside as that has broken at some point, and can probably be looked at as part of fixing that.

closed 2021-10-05 by pfeerick