EdgeTX/edgetx #435: inverting states and switches

jogybaer0815, 2021-07-12:
TX16S c137a1F

For <LOGICAL SWITCHES> and <SPECIAL FUNCTIONS>:
 please add an 'invert <!>' to the selection menu <switch>.
This would make the assignment much easier, as you don't have to scroll through the long lists to select an inverted func or switch e.g. a '! L02'

pfeerick, 2021-07-12:
As in, you could pick SAup and then just click/touch something to use the inverted state... same for L02, FM0, etc?

jogybaer0815, 2021-07-12:
Yes. 
Ideally, the selection menu (SAup, etc.) could already contain the point invert (!) and clicking inverts the existing one.
Second possibility: long click inverts the existing switch SAu ->! SAu and vice versa

pfeerick, 2021-10-03:
Implemented by #594

closed 2021-10-03 by pfeerick