EdgeTX/edgetx #437: Invert channels in the mixer menu

Mantamatt1, 2021-07-12:
Just a suggestion,  when setting up a model I'm spending most my time in the mixer menu. Can we have a way to invert channels within the mixer? Rather than going to outputs.
I know you can change 100 to -100 but be cleaner to have a check box

lshems, 2021-07-12:
There is a shortcut to invert weight on some radios. Plus and minus at the same time

lshems, 2021-07-12:
And if you are flying planes, it is really convenient and recommendable to keep mixers to output what you want the plane control surfaces need to do. So flap down is mixer for flap negative, rudder left is mixer rudder negative, and so on.

Do the inversion part on the outputs.

It will save you a lot of headaches on flaps, ailerons, vtails, all.

Mantamatt1, 2021-07-12:
Ok I get what you mean. Yes it could get confusing. Ok leave as it is

lshems, 2021-07-12:
I'm not deciding. Just trying to challenge... Lol.

It would be nice to have a list of all these kind of key shortcuts. Plusminus at the same time, go figure. For roller radios I don't even know if it exists.

lshems, 2021-07-12:
But if you think it's ok after my challenge, indicate it by closing your request with why you think it can be closed.

Normal ethics I follow:

> Don't close other people's issues
> Challenge requests you feel are solved in other ways. This helps keeping a consisten functionality, good for all
> But also challenge the challenge, especially when the original issue is aiming at making things more user-friendly.

Just my ideas on how it should work. And last thing, don't stop posting! Even stupid looking things or requests for magic might turn out useful or implemented before you know.

Mantamatt1, 2021-07-12:
You made a good point and if the devs challenge me I'm going to back down.
I'm just an end user reaping the benefits of your hard work and devotion. I
just fly expensive toys, who am I to argue?

The main point was to be able to change the servo direction of the output
from the mixer page.

I'm just trying to steer the direction on usability and make a bigger gap
from opentx.

Too many comments asking to make edgetx like opentx because that's what
they are use to. I dont want that as I found opentx too clunky on colour
tx's

pfeerick, 2021-07-13:
I'm on the fence on this one... at the moment, if there is a need to reverse a channel, you just go right to the outputs page... yes, you could reverse input or mix, but generally you make sure the inputs are the right way around, do whatever mixing you need, and then set any limits,  center point or reversing, etc on the output page for the channel. If you have multiple lines for a channel, in the mix, doing it this way would also mean you have to edit each line. 

The current way also means the you can see the bars moving in opposite directions in the channel monitor (as well as a icon shown), so you know at a glance which channels are 'reversed'.

I'm not saying I'm against the idea, just that *I* think it needs more discussion.

gagarinlg, 2023-03-18:
stale

closed 2023-03-18 by gagarinlg