EdgeTX/edgetx #439: Widgets to reflect box size

Mantamatt1, 2021-07-12:
Tx16s
Rc3
If I put image in a small box nothing is displayed. Not even model name. Also if I put the timer in a larger box it is still the same size as in a smaller box, be nice to have a larger timer so I can see it better at a glance
Same applies to other widgets

raphaelcoeffic, 2021-07-13:
Some built-in widgets would need some tweaking, indeed. Could you please make some screenshots of which “small box” you are writing about?

pfeerick, 2021-07-13:
I'm guessing ModelBmp is the "image" one... and when anything smaller than a 1x2 layout is used, the model name isn't shown, just the model image. Meaning if you don't have an image at all, it will just be a white box. 

i.e. each of the white boxes that follows is a ModelBMP widget that did work, until I deselected the image. 

[image]
[image]
[image]

Timer certainly needs to get bigger. i.e. timer is perfect in a 4x4 and also smaller I think, as it's fine as a top bar widget, but doesn't get any bigger if you have have it in say a 1x2 layout (or the bigger of the 2+1 layout).

Mantamatt1, 2021-07-13:
Exactly that

offer-shmuely, 2021-12-15:
The widget **_"Timer2"_** is more aware to to the box area 
so the size is more adptive

https://github.com/EdgeTX/edgetx-sdcard/tree/master/sdcard/horus/WIDGETS/Timer2

pfeerick, 2021-12-15:
Indeed. I have no immediate plants of making any changes to the built in Timer widget since Timer2 is much better at adapting.

offer-shmuely, 2022-01-01:
I think that the best plan should be to move all(!) the widgets to LUA, (thing that currently still cannot be done)
this way the LUA API will match more rich, and the widget will be more customizable

pfeerick, 2022-01-01:
There will probably always be a few widgets (basically just the ones that exist now) that will be baked into the firmware, but no major changes to them as Lua can generally do a better job or be more flexible in allowing changes to be made. I think there are just one or two minor modifications needed and then I think they will probably remain as they are. I reserve the right to be proven entirely wrong as there are some major GUI backend rework efforts afoot! 😁

offer-shmuely, 2023-04-01:
We have a larger timer, 
So can this be closed?

pfeerick, 2023-04-02:
I think so - as I mentioned before I have no intentions of altering the built in timer, and no-one else has stepped up to do so. Plus, Timer2 is a perfectly valid alternative if you want a larger timer, especially since Lua widgets can override built-in ones if given the same name ;)

closed 2023-04-02 by pfeerick