EdgeTX/edgetx #441: Loss of Chinese characters

norulers, 2021-07-13:
[image]
[image]

**General information**
 - EdgeTX version (2.4.0 RC4)
 - Hardware (T18)

raphaelcoeffic, 2021-07-13:
This quite unfortunate, but the biggest fonts do not include the localized characters. This is more or less on purpose to avoid blowing the flash memory up. The best fix would be to use a smaller font.

MervDale, 2021-07-19:
Would not blowing up the flash memory be also an issue of the huge number of characters need in some languages even despite using a smaller font?

raphaelcoeffic, 2021-07-19:
> Would not blowing up the flash memory be also an issue of the huge number of characters need in some languages even despite using a smaller font?

ATM, only the characters used in translations are included in the font, thus massively reducing the size of the fonts. But there are still quite a few left.

richardclli, 2021-08-06:
> 
> 
> This is quite unfortunate, but the biggest fonts do not include the localized characters. This is more or less on purpose to avoid blowing the flash memory up. The best fix would be to use a smaller font.

But if you only include those characters that is required in big font, then it will not blow up memory.  I think in general,  both translations size should be used to select what characters are required and included during build.  I think this will works.

raphaelcoeffic, 2021-08-06:
> But if you only include those characters that is required in big font, then it will not blow up memory. I think in general, both translations size should be used to select what characters are [...]

We cannot afford to have a mapping that depends on font size, really, nor can we safely detect which characters are used in which font size.

richardclli, 2021-08-14:
Or can we make use of images instead.

richardclli, 2021-12-16:
Chinese problems:
- character missing, wrong encoding, etc.

Will work on a fix.

gagarinlg, 2022-09-17:
@richardclli is this still a problem in the current nightlies?

closed 2022-10-16 by raphaelcoeffic