EdgeTX/edgetx #442: Lua function getValue(string) returns zero for sensor max. e.g. "Alt+"

jfrickmann, 2021-07-13:
I have a model with altimeter, and while the sensor "Alt+" reports the correct value on the X9D+ radio's telemetry screen, the Lua function getValue("Alt+") always returns zero.

raphaelcoeffic, 2021-07-13:
Is this only with vario? Or are other sensors involved? And, is this only with LUA, or in general?

jfrickmann, 2021-07-13:
I have not tested other sensors yet

pfeerick, 2021-07-13:
With Alt/Alt+/Alt- (coming from a FrSKY variometer sensor) and RSSI/RSSI+/RSSI- value can see all three, widget can't see + or - .

[image]

```
local defaultOptions = {
}

local function create(zone, options)
  return { zone=zone, options=options, event = 0 }
end

local function update(widget, options)
  widget.options = options
end

local function background(widget)
end

local function refresh(widget, event)
  altValue = getValue("Alt")
  altPlusValue = getValue("Alt+")
  altMinusValue = getValue("Alt-")

  lcd.drawText(1, 0, "altValue: " .. string.format("%.2f", altValue))
  lcd.drawText(1, 15, "altPlusValue: " .. string.format("%.2f", altPlusValue))
  lcd.drawText(1, 30, "altMinusValue: " .. string.format("%.2f", altMinusValue))
end

return { 
  name = "altValue", 
  options = defaultOptions, 
  create = create, 
  update = update, 
  refresh = refresh, 
  background = background
}
```

[image]

```
local defaultOptions = {
}

local function create(zone, options)
  return { zone=zone, options=options, event = 0 }
end

local function update(widget, options)
  widget.options = options
end

local function background(widget)
end

local function refresh(widget, event)
  rssiValue = getValue("RSSI")
  rssiPlusValue = getValue("RSSI+")
  rssiMinusValue = getValue("RSSI-")

  lcd.drawText(1, 0, "rssiValue: " .. string.format("%d", rssiValue))
  lcd.drawText(1, 15, "rssiPlusValue: " .. string.format("%d", rssiPlusValue))
  lcd.drawText(1, 30, "rssiMinusValue: " .. string.format("%d", rssiMinusValue))
end

return { 
  name = "rssiValue", 
  options = defaultOptions, 
  create = create, 
  update = update, 
  refresh = refresh, 
  background = background
}
```

jfrickmann, 2021-07-13:
@pfeerick you nailed it - thanks!

closed 2021-07-14 by raphaelcoeffic

jfrickmann, 2021-07-14:
I works beautifully now - thanks!

commit 6d11fa80 (2021-07-14): [lua] Fix sensor max and min matching

Resolves #442