EdgeTX/edgetx #443: Instant trim - trimming further than trim range

pfeerick, 2021-07-13:
RM TX16S, 2.4.0RC4

hawkeyed on Discord reported that Instant Trim is acting weirdly:
> Just a small issue with instant trim on the Eachine TX16S once activated if you want to move those trims they will only move in one direction until you get full wrap  around then they operate in both directions as per normal,  not sure if you are aware of this.
https://discord.com/channels/839849772864503828/839849772864503832/864481227993120799

My own testing suggests what is happening is if you instant trim outside of the trim range (i.e. trimmed the elevator to 50% off center when trims can only move to 25% off center) it gives the effect that the trims only work one way, since they are actually out of range. You can trim back towards being within the valid trim range, but not further. Because of this, the trim bar has the appearance of wrapping around before it works - once it jumps to the other end of the trim bar you can move in both directions normally - and normal bounds checking will work.

raphaelcoeffic, 2021-07-14:
is that actually all of it? ed59ce281934b

pfeerick, 2021-07-14:
lol... I'll let you know in the morning. 

The simple test - with normal (not extended) trim range, pull up the channel monitor, pull sticks hard to the corner, and insta-trim... when you let go of the stick, the channel should only be trimmed to 25% off center. Otherwise, instratrim has overtrimmed ;)

pfeerick, 2021-07-14:
On https://github.com/EdgeTX/edgetx/commit/a268a9a1d763d09945303ee4eef63c4728b0ead3, which includes the above commit... instant trim is still setting the trims out of bounds, and now the trim indicator knob/box is off-screen.

raphaelcoeffic, 2021-07-15:
Ok, I’ll check again. Maybe I missed something. The trim steps are constrained by the range (depends on wether extended was set or not), but maybe there is something I missed.

pfeerick, 2021-07-15:
Well, something here must be forcing it out of range to start with...
https://github.com/EdgeTX/edgetx/blob/64f277152cd87d8c72e13583c092d3061a44fa8c/radio/src/opentx.cpp#L1555-L1593

Maybe the setTrimValue() on 1590 needs a kick (I've not looked)... Dinner ;)

raphaelcoeffic, 2021-07-15:
Ok, a couple of findings which should shed some light on how it is intended:
- B/W targets display trims using clipping, so that extended trims show-up the same once they went over the "normal" limit.
- Instant trim allows for full range (i.e. extended trims), the code is the same on opentx.
- once a value is outside the allowed window (extended trims NOT configured), you can only go back with the value, which is intended (extra code for that).
- audio feedback only accounts for normal range, meaning that once you get over the fence (into extended territory), the audio freq does not increase/decrease further, and you don't get the announcement once you hit the real extended limit. That should be fixed.

raphaelcoeffic, 2021-07-15:
> So the question is... should instant trim be able to set the trims beyond the trim range?

It trims within the extended range, no matter what. It really looks intended in the code, as there is special casing for returning back to configured range, without allowing to trim further outside the configured range.

That being said, at no time the trims cross the boundaries of extended trims.

pfeerick, 2021-07-15:
Ah, yes, I saw that in the code... so mostly realized, but went to dinner 😆 

So, that makes sense, and the trim behaves exactly like that on the X9D+, especially when you turn on the extended trim. in that case, I would like to a) grouse about the OTX sim not replicating that behaviour (don't have OTX on a colour screen TX atm, and didn't want to swap back again for like the third time today...) ... it limited the insta-trim to within the trim range, rather than extended limt and b) say that that commit is probably all that's needed... just need the trim box to stay sensibly in view! 😆

raphaelcoeffic, 2021-07-15:
> just need the trim box to stay sensibly in view! 😆

PR #454 takes care of that, so we should be good. From what I tested, it looks good under all conditions.

MervDale, 2021-07-15:
Just in relation to the Instant trim issue the problem is that it is not consistent and not always repeatable

raphaelcoeffic, 2021-07-15:
> Just in relation to the Instant trim issue the problem is that it is not consistent and not always repeatable

Could you please try the firmware corresponding to this PR (see "Checks"/"Artifacts") ?

pfeerick, 2021-07-15:
> it is not consistent and not always repeatable

In what way? i.e. In really, what Instant Trim is really doing is using the trims to change the output when the stick is in the neutral position to be the same as where you currently have the stick pushed over... so if you push the stick over a bit, and keep retriggering instant trim, it will keep moving that centre point. Before the PR in question, if you had the stick pushed over more than the 25%+/- movement allowed by trims by default, it would let the UI go a bit crazy, but the trims were still working as expected, as they were working within the 'extended trims' range. Ideally, if you needed to trim that severely, you would had already enabled that first, so you have full control over the trims even when pushed that far. 

One way to see this in action is to push the stick over a bit, and keep re-triggering instant trim... whilst also having a channel output widget on screen. You'll see the output bar creep over each time you trigger instant trim, as well as the trim knob move... just before it could go off screen, or too far. Now it will go yellow to indicate it's gone into extended trims territory, and will stay put.

commit cdbc62c5 (2021-07-15): [color] clip trim value displayed based on range set.

Resolves #443
closed 2021-07-15 by raphaelcoeffic