EdgeTX/edgetx #444: Accidentally altering a checkbox

RC-SOAR, 2021-07-13:
2.4.0, RM TX16S

Nice job with the UI, howver I found it too easy to alter a checkbox accidentally when swiping.

Maybe add an 'edit' state for checkboxes:

- Press once to enter 'edit' mode (marked with green border)
- Press again to toggle the on/off state. 
- Press elsewhere to cancel edit mode.

jogybaer0815, 2021-07-13:
This should be prevented by a timing latency, but that is not easy to implement

raphaelcoeffic, 2021-07-14:
> This should be prevented by a timing latency, but that is not easy to implement

Especially, it would require some serious fine tuning to be "blurry enough" when needed, and "sharp enough" when required. But someone comes up with a clever trick, I'll take it.

pfeerick, 2021-07-14:
Just don't press them! :-P I usually scroll on the left side of the screen to avoid any input elements. 

But in all seriousness, I'd be curious to know the specific movements that were done to actually the checkboxes accidentally... as I am actually finding it hard to do even deliberately... ie. I'm pressing on a checkbox and swiping, and it's not being toggled. It only toggles if I tap on it first... which is what I would expect.

NemesisXB, 2021-07-14:
Part of the problem is that you don't even have to tap directly on the checkbox. If you tap all the way on the right edge of the screen in line (horizontally) with the checkbox, you can still check and uncheck it. So basically touching the screen on the checkbox and anywhere to the right of the checkbox still checks/unchecks it. Maybe this should be registered as a separate issue?

pfeerick, 2021-07-14:
Oher... nice one! That is a pain! 😮

Mantamatt1, 2021-07-18:
I noticed it's been happening to me also. I was looking through the radio setup and realised every slider was on max. I.e volume haptic frequencys etc. 
As I'm sliding with my right thumb I didnt realise I've been turning everything up. I'll try using my left thumb for scrolling. I'll probably start to turn everything to the minium then lol

pfeerick, 2021-07-18:
Just keep alternating between left and right thumb, and that should balance it up again ;)

jogybaer0815, 2021-07-18:
there are special reasons why the checkboxes show this behavior? 
Is there only the height of the checkboxes for touch coded?
is the query box + edge too complex to program?
It is good for the user to know if there are these restrictions.
EdgeTX has many great features that are worthwhile...

NemesisXB, 2021-07-18:
A fix for the slider is to somehow only allow actually sliding the marker and not allow it to jump if you randomly tap somewhere on the line.

RC-SOAR, 2021-07-18:
Thanks for the thoughtful responses. Having to debug a broken checkbox - or any other UI element - in the heat of a competition is not a nice prospect.

pfeerick, 2021-07-19:
> A fix for the slider is to somehow only allow actually sliding the marker and not allow it to jump if you randomly tap somewhere on the line.

I think the checkboxes have this behaviour because they span a whole panel. I think that for sliders, it would be nice to be able to either just grab the knob and drag it, or touch it once to select the slider, and a second time to jump to where you are touching. For the checkboxes, I think either the proposed, or making it so *only* the checkbox is the active touch zone, not the rest of the grid element it occupies.

Please do not mistake my tongue-in-check responses for dismissal of the issue... that would have been a closing of the issue and wont-fix label applied. It's just not on _my_ todo list yet, and we still juggling at the moment the priority of what needs to be worked on.

eshifri, 2021-07-19:
If I read the code correctly, CheckBox::rect may be much bigger than visible checkbox. So the solution is to overwrite the CheckBox::onTouchEnd() and shrink the area accepted by this function. 
I have submitted the pull request for libopenui.

RC-SOAR, 2021-07-21:
These are welcome improvements but ultimately still just mitigations. 

Let's accept that certain UI elements are less than secure - and design around it.

The Mixers and Outputs already show the way with a context menu for editing (great!!). Could you not extend this to the Model Setup, Flight Modes and Heli Setup screens? That way, _all_ the main screens would be locked by default.

I don't want to make a big issue of this, but personally this is the least I would consider for a competition-capable rig.  Even the slightest risk of accidental changes when browsing reduces confidence in the UI... or worse.

NemesisXB, 2021-07-21:
> These are welcome proposals but ultimately still just mitigations.
> 
> Let's accept that certain UI elements are less than secure - and design around it.
> 
> The Mixers and Outputs already show the way with a context menu for editing (great!!). Could you not extend this to the Model Setup, Flight Modes and Heli Setup screens? That way, _all_ the main screens would be locked against edits.
> 
> I don't want to make a big issue of this, but personally this is the least I would consider for a competition-capable rig. Even the slightest risk of accidental changes when browsing reduces confidence in the UI... or worse.

Still, if you have a sub-menu that spans more the the screen height, you still need to scroll and again, you can edit the checkbox or slider by accident. I understand what you want and why. I not sure everybody will agree on this. UI design is very subjective 

But I think your request is a different issue and as such you should create another issue. This issue is to try and prevent accidental changing of settings while scrolling.

RC-SOAR, 2021-07-21:
> Still, if you have a sub-menu that spans more than the screen height, 
> you still need to scroll, and again, you can edit the checkbox or slider by accident. 

That's okay, because it's a sub-menu for editing.  I click the 'edit' option, and I accept the risk. 

What is not acceptable is _any_ risk when casually browsing the main screens.

For me, this discussion has clarified a couple of key points:
1. Checkbox risks can be mitigated, not eliminated
2. We need to look at the menu design for a complete solution. And yes, I can raise a separate issue.

commit 5a272cbf (2021-07-31): Update libopenui to include a fix for checkboxes

Resolves #444
closed 2021-07-31 by raphaelcoeffic