EdgeTX/edgetx #445: Bug Logical Switches Function Timer

yak55, 2021-07-13:
Bug Logical Switches

Function e.g. greater than ... but also all other functions
V1 = timer - no matter which one
V2 = no entry possible !!

pfeerick, 2021-07-14:
Repeatable, TX16S. Should editable time ... i.e. HH:MM:SS
Does not effect X9D+ or T-Lite.

yak55, 2021-07-14:
Hello Peter, thanks for answering.
My radio is Radiomaster TX16S.
Function  works correct in Open TX.
Not so in Edge, logical switch entry in V2 is deleted.
No new entry is possible again.
Input box is blocked. (Green line )
Photo is attached
Best regards Eugen
[image]

eshifri, 2021-07-16:
This functionality is not implemented: see draw_functions.cpp line 439

pfeerick, 2021-07-16:
But hey... at least there's a TODO comment! 😆 

Poor innocent TODO comment... it's been waiting a whole two years for some love...

https://github.com/EdgeTX/edgetx/blob/2599c773e282c151c4083c94870f7189f09f5cf7/radio/src/gui/colorlcd/draw_functions.cpp#L440

closed 2021-07-18 by raphaelcoeffic