EdgeTX/edgetx #458: DSM Channel order

Glennb007, 2021-07-15:
Some issues were listed but not directly related. Comments on the forums are there but no results.

Binding DSM receivers results in sometimes binding but in the wrong channel order. Aileron is the elevator, the rudder seems always OK but the motor either works or doesn't. Even when it does bind, when you turn it back on it goes nuts again. Hit or miss. Seems to be only the receivers with AS3x

There is nothing wrong with the receivers as it will bind with my Spectrum transmitters instantly and works correctly. These were all working with Open TX. 

Using Edge TX 2.4 RC4
All DSM receivers. Transmitter Radiomaster DX16S. Planes with receivers built-in include UMX Timber, UMX Carbon Cub SS. Spectrum AR635 used in a Parkzone Visionaire.

pfeerick, 2021-07-15:
So these would all be DSM2 receivers? and using the internal Multi Protocol Module?

I believe there is an issue with the default settings for DSM, and since some of the options are currently missing for DSM, this is where things get unstuck. Particularly, I think the 22ms servo rate is being set, on a 11ms link (2 11F)... and that just ain't right! I had more luck with DSMX, but it was a little unstable, so still something wrong in default settings for it. 

Regarding the channel mapping, what order do you have the channels in your mixer? And what order does your MPM expects to see them in? Please ignore the outputs widget if you're using that for reference, as it has a bug (that has been fixed post RC4) that will only confuse you more. Open up the channel monitor, and check what channel moves when you move a stick. 

i.e. I have my MPM programmed to expect TAER order, and have my MIXES in TAER order -- and it is working for me with DSM2/DSMX. However, if the channel remapping option is disabled (another option currently missing), and the MPM is in AETR order this would explain what is happening (as this tells the module to remap it's channel order to the order the protocol in use expects, where relevant).

pfeerick, 2021-07-16:
Jesus, github is stupid when trying to assign milestones. Anyway, this is also related to #429

Glennb007, 2021-07-16:
Channel order is AETR. Also, have the latest version of the 4 in 1 module.
All are DSMX. I checked the monitor before I posted and everything is correct.

pfeerick, 2021-07-16:
If you have AETR all the way through (from channel mapping to module - the default for mine was AETR but I changed it to TAER), and the receiver is wrong... we can be pretty sure it's that the channel remapping is disabled by default atm! 😢 

[image]

Glennb007, 2021-07-16:
Mine is V1.3.286 AETR. Is anyone going to fix this or just let it slide?

pfeerick, 2021-07-16:
Nah, we thought we'd let everyone crash and get better receivers! 🤦‍♂️ 

Of course it will be fixed. The first step is identifying exactly what is needed to reproduce the issue. So accordingly, I'll reflash my MPM to back to AETR, and chance my mixer channel order to AETR also, and I should then have the channel mixup issue as well. Then can see if it is just a matter of toggling the MPM flags so it does the remap. If so, once the GUI stuff options are back in (hopefully shortly), it should be right. 

I can't make any promises if it will make 2.4.0, or be a nightly shortly after that. I for one won't have much time the next couple of days to look at it, and may only have time to confirm what changes are needed over the weekend.

olliw42, 2021-07-16:
I can't say I really understood what's been discussed here, but what I want to throw in is that in my observation there is at least no difference between ETX and OTX concerning this ... which would imply it's not a specific thing

I btw do use AETR both in the def channel order and MPM firmware, and I had to do a weird shuffle in the inputs for a DSMX receiver - which is what you are probably talking about - but I did NOT have to do that for a satellite receiver !!! Maybe not surprising, but I just thought I mention, since this might have been how it may have crept in.

olliw42, 2021-07-16:
>  once the GUI stuff options are back in (hopefully shortly)

I hear you :) I plan to do it tomorrow ... (week was busy)

pfeerick, 2021-07-16:
> AETR both in the def channel order and MPM firmware

The important thing here is actually the mix/output order, not the default. What is *supposed* to happen is *if* you have the channels output in the same order as the MPM expects (i.e. AETR), but you are using protocols/systems like DSM, which expects the output order to be TAER, the MPM is *supposed* to remap that order itself, **as long as the 'Disable Ch. map' option isn't ticked**.

Because I reprogrammed mine right from the start to use TAER, and use TAER consistently all the way through, I never noticed this. I've reprogrammed it to AETR, swapped to AETR outputs, and under ETX, my DSM2 receiver is getting no throttle input, but the remap is happening... i.e. channel three for spektrum is E, but transmitter is outputting E on channel two...

olliw42, 2021-07-16:
> What is _supposed_ to happen is _if_ you have the channels output in the same order as the MPM expects (i.e. AETR), but you are using protocols/systems like DSM, which expects the output order to be [...]

yes, I understand that well :)

the point I wanted to make is that ETX and OTX have both the same behavior ...

pfeerick, 2021-07-16:
I'm not seeing that. With DSM2 11F, switching back to OTX 2.3.13 and 2.3.11 on the same TX16S results in the channel remapping still working, but no servo movement on CH1 or CH4 of the DSM2 receiver (CH2 and CH3 are working). On ETX, CH4 works as well, leaving only the throttle non-functional. On both cases, there is *no* output on the non-functional channels - it's not varying, and it's not channel center. So there is clearly something *very* weird going on here. I'm starting to wonder if I should revert the version of my MPM, maybe it's a bug in the newer firmware as they did do extensive re-writes recently and were warning not everything was tested...

pfeerick, 2021-07-16:
Taking this a bit further, changing the MPM firmware made no difference, so back on the latest AETR version. However, I have noticed that a single valid packet must be sent out for the throttle, as my receiver tester was registering a change - and it will vary depending on what position the throttle is in when the transmitter turns on - i.e. when it first talks to the receiver. but it never changes again. I've now checking with a oscilloscope, and when the transmitter is powered on, or you switch to the DSM2 module, there is a single pulse sent out of the throttle channel of the receiver, when it should be continuous. 😢

Glennb007, 2021-07-16:
All was working well under Open TX.

Lenb78, 2021-07-17:
Hi just to add my issue. I set my tx16s set as TAER as I am converting from Spektrum. it says when I bind as ATER. but the when check the channels on the reciever if you go ch1 which on spektrum is throttle its elevator. then ch2 “Ailerons”throttle, then ch3 “elevator” is Ailerons. and ch4 is correct Rudder. 
Not the end of the world for a new model. just a pain for older models with gryros.

pfeerick, 2021-07-17:
Can I just double check, so the order of the channels in your Mixes page are TAER in your Mixes page? As the channel ordering only applies to newly made models, so just need to double check it was applied.

Glennb007, 2021-07-17:
Went back to Open TX. From what I remember it was AETR on all. I tried
setting up a new model but got the same results. Thought it might be in the
conversion process but I guess not.

Lenb78, 2021-07-17:
> Can I just double check, so the order of the channels in your Mixes page are TAER in your Mixes page? As the channel ordering only applies to newly made models, so just need to double check it was applied. 

Correct TAER on transmitter.
reciever shows as ETAR

Lenb78, 2021-07-18:
thanks guys works like a treat!!![image]

hit disable channel map and bang TAER!!

raphaelcoeffic, 2021-07-18:
@Lenb78 the options were mislabeled. I believe it should correct now.

pfeerick, 2021-07-18:
I didn't get time to sir down and figure out the channel reordering, but that sounds right - if you put your mixes in ATER order for that model it should also work properly when you untick the disable option (and should have before unless something was really broken).  

What you may want to do at some point is reflash the MPM in TAER channel order, for any of various bind and fly models that might want a different order which the MPM will work out. If you have the channels in the same order the MPM wants, it won't try to remap at all ;)

Thank you for testing!

Glennb007, 2021-07-18:
The fix got the controls to working correctly but the throttle does not work. Tried on 2 planes. Tried reflashing the 4 in 1 to TAER with no flashing. Says it is not the right file although it is a bin file. Any suggestions? Thanks

raphaelcoeffic, 2021-07-18:
> The fix got the controls to working correctly but the throttle does not work. Tried on 2 planes. Tried reflashing the 4 in 1 to TAER with no flashing. Says it is not the right file although it is a bin file. Any suggestions? Thanks

Can you tell if the ESC arms properly? It might need an even smaller value as `0`-position to arm properly? (maybe "extended limits" + "extended throw"?)

Glennb007, 2021-07-18:
It arms properly as I get the 2 beeps (2s) and bind tone. Enabled max through but no difference.

Glennb007, 2021-07-18:
> > The fix got the controls to working correctly but the throttle does not work. Tried on 2 planes. Tried reflashing the 4 in 1 to TAER with no flashing. Says it is not the right file although it is a bin file. Any suggestions? Thanks
> 
> Can you tell if the ESC arms properly? It might need an even smaller value as `0`-position to arm properly? (maybe "extended limits" + "enable max. throw"?)

It arms properly as I get the 2 beeps (2s) and bind tone. Enabled max throw but no difference. UMX Timber and UMX Carbon Cub SS. Going to try non UMX and see what happens

Glennb007, 2021-07-18:
> It arms properly as I get the 2 beeps (2s) and bind tone. Enabled max throw but no difference. UMX Timber and UMX Carbon Cub SS. Going to try non UMX and see what happens

Update: non AS3X receivers work fine. Have full function.

Glennb007, 2021-07-18:
Correction: Spectrum AR635 with AS3X works fine now. Have all controls. Maybe this is an issue with UMX boards with software although some of my UMX planes that do not have AS3X works fine. UMX Radian and ASK21 and UMX J-3 (3 channel)

pfeerick, 2021-07-19:
Did they work before? Just wondering if there is anything fancy that needs to be done to them like using the [CH14 throttle kill](https://www.multi-module.org/using-the-module/protocol-details/dsm#throttle-kill).

Glennb007, 2021-07-19:
No on the AR635. Controls were wrong. Aileron was the elevator. The rudder was fine, No throttle. Works fine now. From the planes that don't work, they all have AS3X and UMX. The AR635 does have AS3X it is OK. Hope to get the UMX problem solved as I do fly them at my house on 7 acres. Big planes at our club.

pfeerick, 2021-07-19:
Ok, so the order problem _may_ have been the do to with channel remapping unless you'd verified on the models that were out of order that the channel order on the MIXES page for _those specific models_ was AETR since your MPM expects AETR if channel remapping is not disabled. 

Is this one of the ones that isn't working? So it's just a Spektrum DSM receiver + their AS3X stabilisation system? I'm surprised it wouldn't be working since it's DSM2 and DSMX compatible (although they recommend DSMX 22ms), so there shouldn't be much to getting it to work really. I might leave this issue closed, but feel free to comment further, and can open another issues if we can isolate it to ETX and not something to do with setup or MPM generally. 

i.e. If the model worked on the same transmitter with OpenTX, but not EdgeTX... we're doing *something* differently.

closed 2021-07-19 by pfeerick

Glennb007, 2021-07-19:
> Ok, so the order problem _may_ have been the do to with channel remapping unless you'd verified on the models that were out of order that the channel order on the MIXES page for _those specific models_ was AETR since your MPM expects AETR if channel remapping is not disabled.
> 
> Is this one of the ones that isn't working? So it's just a Spektrum DSM receiver + their AS3X stabilisation system? I'm surprised it wouldn't be working since it's DSM2 and DSMX compatible (although they recommend DSMX 22ms), so there shouldn't be much to getting it to work really. I might leave this issue closed, but feel free to comment further, and can open another issue if we can isolate it to ETX and not something to do with setup or MPM generally.
> 
> i.e. If the model worked on the same transmitter with OpenTX, but not EdgeTX... we're doing _something_ differently.

ALL models (16) were working fine on OpenTX. Only when switching to EdgeTX did the DSM problem start. 70% of my receivers are DSM. Thanks

Glennb007, 2021-07-23:
Just an FYI. All is working now on DSM UMX bricks. You must change the number of channels to 6. The default is 8. The throttle started working on a hunch I had.  Thanks again for the help.

PR #471 (2021-07-19): [color] fixed MPM options (esp. DSM related)
Resolves #458 and possibly other related MPM issues.