EdgeTX/edgetx #464: ELRS [XF] CRC error

NemesisXB, 2021-07-16:
**Issue**
While testing the touch screen lockup with a debug build I caught a bunch of [XF] CRC error messages.
Here is an extract log where it sustained the error for a very long time.
[XF_CRC_Error.log](https://github.com/EdgeTX/edgetx/files/6832449/XF_CRC_Error.log)
It happens quite often. Mostly not as extended:
[XF_CRC_Error_short.log](https://github.com/EdgeTX/edgetx/files/6832512/XF_CRC_Error_short.log)

**General information**
 - EdgeTX version (8bfbb723)
 - RM TX16S
 - Happymodel TX2400
 - ELRS 1.0.0-RC9

raphaelcoeffic, 2021-07-17:
@rotorman so you think this is related to your changes? Normally this does not happen.

NemesisXB, 2021-07-17:
> @rotorman so you think this is related to your changes? Normally this does not happen.

If you give me a binary with debug output without @rotorman touch changes, I can test this.

rotorman, 2021-07-17:
> @rotorman so you think this is related to your changes? Normally this does not happen.

I do not believe so. This is likely a independent issue.

@NemesisXB here you go with TX16S firmware of the current 2.4 head (w/o PR #457 ), built with -DDEBUG=YES.
[20210717 1016 EdgeTX 2.4. head TX16S DEBUG out.zip](https://github.com/EdgeTX/edgetx/files/6834401/20210717.1016.EdgeTX.2.4.head.TX16S.DEBUG.out.zip)

pfeerick, 2021-07-17:
lol... snap... I was about 30 seconds up uploading also!

NemesisXB, 2021-07-17:
> I do not believe so. This is likely a independent issue.
>
> @NemesisXB here you go with TX16S firmware of the current 2.4 head (w/o PR #457 ), built with -DDEBUG=YES.
> [20210717 1016 EdgeTX 2.4. head [...]

Still Happens. Ran this one for about 450 seconds happened once (every time it happens, it happens a couple of consecutive times). Ran the one with the PR #457, and it happened 3 times. See attached image. 1016.log is with PR #457 and 1444.log is without also ran it for 450 seconds. Without PR #457 I ended with 58 TSI2CEvents and with PR #457 ended with 1.
[image]

NemesisXB, 2021-07-17:
After some more testing it definitely seems to happen more on the version with PR #457 than without. I ran both versions. This time touching the screen as little as possible. The results are:

rotorman, 2021-07-17:
PR #457 briefly disables all interrupts and then enabling them back, as it is clearing the I2C ADDR bit (to atomically read two registers SR1 and SR2) as it is processing I2C stream:
https://github.com/EdgeTX/edgetx/blob/5f0bac3dc94a9ee3f4307c4a1dea6dacbd5787b0/radio/src/targets/horus/i2c_driver.h#L234-L242

The original STM32F4 HAL implementation does not actually disable/enable interrupts, so we could test if by removing this makes it better here and/or worse for I2C. I will update PR #457.

rotorman, 2021-07-17:
@NemesisXB here is a new binary for TX16S with DEBUG output of the latest PR #457 code that does not disable interrupts when clearing I2C ADDR bit. Would you be so kind and give this a whirl with ELRS. Thanks!
[20210717 1459 EdgeTX TX16S PR427 no IRQ disable while clearing ADDR.zip](https://github.com/EdgeTX/edgetx/files/6834733/20210717.1459.EdgeTX.TX16S.PR427.no.IRQ.disable.while.clearing.ADDR.zip)

raphaelcoeffic, 2021-07-17:
@NemesisXB no clue if this is important, but I just tried with ELRS 1.0.0 (final), and it seems to work for me under all considerations (no touch event, no lost packets, no CRC errors).

NemesisXB, 2021-07-17:
> @NemesisXB no clue if this is important, but I just tried with ELRS 1.0.0 (final), and it seems to work for me under all considerations (no touch event, no lost packets, no CRC errors, packet count stable @ 500Hz).

@rotorman Still happening on my setup.

@raphaelcoeffic, I did upgrade ELRS to 1.0.0. Still happening. Maybe my Happymodel unit is faulty or something. If somebody can build me the latest OpenTX with debug, maybe I can test and compare to that?

raphaelcoeffic, 2021-07-17:
>Maybe my Happymodel unit is faulty or something

I don't think so, it would not have this kind of effects. I'm probably just not very lucky with this issue.

rotorman, 2021-07-17:
@NemesisXB : > If somebody can build me the latest OpenTX with debug, maybe I can test and compare to that?
Here you go.
[20210717 1831 OpenTX 2.3dev TX16S DEBUG.bin.zip](https://github.com/EdgeTX/edgetx/files/6834988/20210717.1831.OpenTX.2.3dev.TX16S.DEBUG.bin.zip)

NemesisXB, 2021-07-17:
Happening on OTX as well! Any Ideas? Should I close this issue and open it on ELRS?

raphaelcoeffic, 2021-07-17:
> Happening on OTX as well! Any Ideas? Should I close this issue and open it on ELRS?

Probably discuss it with ELRS first.

NemesisXB, 2021-07-18:
Discussed with the ELRS devs. Seems to only happen with the debug build. @raphaelcoeffic @rotorman if you are happy we can close this one.

raphaelcoeffic, 2021-07-18:
> Discussed with the ELRS devs. Seems to only happen with the debug build. @raphaelcoeffic @rotorman if you are happy we can close this one.

Ok, thx, we can live with that.

closed 2021-07-18 by raphaelcoeffic