EdgeTX/edgetx #473: Luascript: Text color within lcd.drawFilledRectangle in title not inverted

jogybaer0815, 2021-07-19:
TX16S nightly release 6eef71d

Text within title with filled rectangle appears not inverted as expected.

code snippets:

 local function drawScreenTitle(title,page, pages)
  lcd.drawFilledRectangle(0, 0, LCD_W, 30, TITLE_BGCOLOR)
  lcd.drawText(1, 5, title, MENU_TITLE_COLOR)
  lcd.drawText(LCD_W-40, 5, page.."/"..pages, MENU_TITLE_COLOR)
end
 
That appears as black (new standard color - previously red) over filled darkblue rectangle :(
my temporary workaround: 

local function drawScreenTitle(title,page, pages)
--  lcd.drawFilledRectangle(0, 0, LCD_W, 30, TITLE_BGCOLOR)
  lcd.drawRectangle(0, 0, LCD_W, 30, TITLE_BGCOLOR)
  lcd.drawText(1, 5, title, MENU_TITLE_COLOR)
  lcd.drawText(LCD_W-40, 5, page.."/"..pages, MENU_TITLE_COLOR)
end

that is readable.
openTX shows inverted text within filled rectangle
greetings

pfeerick, 2021-07-19:
So is the problem using the theme colours, rather than specifying your own?

jogybaer0815, 2021-07-19:
I have only removed the pilot picture from theme

pfeerick, 2021-07-19:
Pilot picture?

raphaelcoeffic, 2021-07-19:
> Pilot picture?

He probably means the default background.

jogybaer0815, 2021-07-19:
sorry for that confusion: yes background.png

JimB40, 2021-11-11:
still valid?
Do we support color contants legacy? I mean fix OTX bugs

pfeerick, 2021-11-12:
> Do we support color constants legacy? 

Yes, unless we want to ensure a lot of old scripts become incompatible, which is not an option just yet IMO. 

However, checking the behaviour of OTX vs ETX, I think this was fixed in v2.5 also through theme constant changes... The unreadable red on dark blue certainly seems to be gone ;)

OTX:
[image]

ETX:
[image]

```
local page = 1
local pages = 3
local firstRun = 0

local function drawScreenTitle(title,page, pages)
  lcd.drawFilledRectangle(0, 0, LCD_W, 30, TITLE_BGCOLOR)
  lcd.drawText(1, 5, title, MENU_TITLE_COLOR)
  lcd.drawText(LCD_W-40, 5, page.."/"..pages, MENU_TITLE_COLOR)
end

local function run(event)
    if firstRun == 0 then
      firstRun = 1
      lcd.clear() 
      lcd.drawFilledRectangle(0, 0, LCD_W, 30, TITLE_BGCOLOR)
      lcd.drawText(1, 5, "Welcome (Run)", MENU_TITLE_COLOR)
      lcd.drawText(LCD_W-40, 5, page.."/"..pages, MENU_TITLE_COLOR)
    end

    if event == EVT_VIRTUAL_NEXT_PAGE and page < pages then
      page = page + 1
      drawScreenTitle("My Title (Next)",page, pages)
    end

    if event == EVT_VIRTUAL_PREV_PAGE and page > 1 then
      page = page - 1
      drawScreenTitle("My Title (Prev)",page, pages)
    end

    if event == EVT_VIRTUAL_ENTER_LONG or
       event == EVT_ENTER_LONG or
       event == EVT_MENU_LONG then
        -- exit script
        return 2
    end

  return 0
end

return {run=run}
```

closed 2021-11-12 by pfeerick