EdgeTX/edgetx #475: The TX16s on Screen Settings when setting the time in the top banner it doubles stamps the time & it doesn't look good.

JWKillTime, 2021-07-19:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

 The TX16s on Screen Settings when setting the time in the top banner it doubles stamps the time & it doesn't look good.

[image]
[image]

eshifri, 2021-07-21:
This is a new problem: it worked OK at least prior to RC1.

eshifri, 2021-07-21:
And it happens with time and timers on all widgets.

eshifri, 2021-07-21:
The problem occurs if Timer or Time are selected as value->timer. In this case two different font sizes are used to output the time.
If widget is created as timer  - not a value - everything is OK. Unfortunately in this case timer name is not available (only index) and Time is not available either.

JWKillTime, 2021-07-24:
Updated to newest Flasher & firmware & all is sorted. Thanks

closed 2021-07-25 by raphaelcoeffic