EdgeTX/edgetx #478: Channel Mapping for Spektrum

Mikey1968, 2021-07-21:
When I was using edge and boy do I like the layout I came across a problem using Spektrum rxs, Theres also Spektrum auto that was missing . But when I went to bind spektrum , none of the channels would work in the right order.  When I went back to opentx and bound the rx using the auto , all the channels were working in order

raphaelcoeffic, 2021-07-21:
> When I was using edge and boy do I like the layout I came across a problem using Spektrum rxs, Theres also Spektrum auto that was missing . But when I went to bind spektrum , none of the channels [...]

The “Auto” feature is not included yet. If the channel order is not correct, you should check the channel binding option. Did you test on latest 2.4, or on RC4? RC4 still has the issue, whereby it is already fixed in latest 2.4.

Mikey1968, 2021-07-21:
I don’t remember what I had , whats 2.4 and RC4  Sent from Mail for Windows 10 From: Raphael CoefficSent: Tuesday, July 20, 2021 9:28 PMTo: EdgeTX/edgetxCc: Mikey1968; AuthorSubject: Re: [EdgeTX/edgetx] Channel Mapping for Spektrum (#478) When I was using edge and boy do I like the layout I came across a problem using Spektrum rxs, Theres also Spektrum auto that was missing . But when I went to bind spektrum , none of the channels would work in the right order. When I went back to opentx and bound the rx using the auto , all the channels were working in orderThe “Auto” feature is not included yet. If the channel order is not correct, you should check the channel binding option. Did you test on latest 2.4, or on RC4? RC4 still has the issue, whereby it is already fixed in latest 2.4.

Mikey1968, 2021-07-21:
Isn't 2.4.0 the same as RC4

raphaelcoeffic, 2021-07-21:
> Isn't 2.4.0 the same as RC4
> […](#)

No, RC4 is a snapshot of what 2.4 was at the time. However, the branch has evolved to integrate more fixes.

Mikey1968, 2021-07-21:
I downloaded the lastest nightly which looks like it might have the fixes.
Will try it tomorrow

Mikey1968, 2021-07-21:
Everything seems to work so you can close the thread. Thanks for the help

closed 2021-07-21 by raphaelcoeffic