EdgeTX/edgetx #481: Prevent accidental changes when browsing certain menus.

RC-SOAR, 2021-07-21:
This is related to #444.

To summarise: there is a (small) risk of accidentally changing the setup when simply browsing some of the menus. 

The degree of risk is subject to a number of factors, including carelessness, finger size, stress in competition and so on. However _any_ risk when simply browsing (as opposed to editing) is unacceptable.

My proposal relates to the Model Setup, Heli Setup and Flight Modes menus. There are two elements to the proposal:

1. Menus should be locked against editing, so for browsing only.
2. A context menu should be provided with an editing option. This would work just like the Mixers, Outputs and other menus which are currently secure (they work very well!).

Any small risk in the editing menus is understood, and acceptable.

pfeerick, 2021-10-03:
Perhaps through a edit/view type toggle? As having submenus isn't going to be practical UI, nor is it going to be very user friendly... i.e. I could see flight modes screen being possibly changed, but not sure about the others. 

Whereas I can see an overlay icon that allows you to lock all input so it's scroll only being quite effective. The question would be if adopting this sort of approach would be should it be remembered across sessions (i.e. persistent), or is it something that is just occasionally needed, so would normally be in edit mode).

RC-SOAR, 2021-10-03:
I agree - the Flight Modes screen uses repeated blocks, so it would be nice if it could have a dedicated Edit menu like other similar screens (Mixers, Inputs etc.)

The Model Setup screen is the odd man out, since it doesn't use repeated blocks...  I think an edit mode would be a good compromise. Maybe not ideal in terms of having a consistent UI , but if that's the price of making it more secure, then it's justifiable. At power up, the screen should be in the 'locked' state, otherwise it would be too easy to forget the last state if the tx has not been used for a long period.