EdgeTX/edgetx #482: Widget setup main view first

Mantamatt1, 2021-07-21:
Tx16s
Rc4
When creating new model can we swap the first selection to main view?
I.e. hit screen settings and the first menus is user interface. I think it should be main view as that's the most important one.
I think that user interface should be the second page to set up as it's not as important as the main page.

raphaelcoeffic, 2021-07-22:
Ideally, we should land on the edit page for the main view that was shown before: starting from main view 2, go directly to editing main view 2.

Mantamatt1, 2021-07-23:
Or.... get rid of user interface tab completely and integrate it in the main views

pfeerick, 2021-07-24:
I suggest not - as if theme support is re-introduced, that is where it goes, and the other element, top bar, is the same for all panels, so doesn't belong there. There could also be other things added there in the future. I would suggest leaving the ordering of the tabs as they are, and simply selecting the active screen's tab by default.

closed 2021-11-27 by pfeerick