EdgeTX/edgetx #484: Bootloader back drop as home screen background

Mantamatt1, 2021-07-21:
Tx16s
Rc4
When in boot loader the back drop looks cool.
 can it be used as a background?

pfeerick, 2021-07-24:
This is the image used for the bootloader - it's basically just tiled to fit the fill screen. So you should be able to use it to make your own 480×272 px PNG image, and shove it into /IMAGES/EdgeTX/background.png

https://github.com/EdgeTX/edgetx/blob/2.4/radio/src/bitmaps/480x272/bootloader/bmp_background.png

rotorman, 2021-07-24:
Here the background in 480x272 pixels:
[image]

Mantamatt1, 2021-07-24:
Thanks for that. I just couldn't find the image in the sd content

Mantamatt1, 2021-07-24:
> This is the image used for the bootloader - it's basically just tiled to fit the fill screen. So you should be able to use it to make your own 480×272 px PNG image, and shove it into [...]

> Here the background in 480x272 pixels:
> [image]
> 
> I originally had the idea to match the bootloader background with the startup splash background in PR #262
> Should we go with it?

Yes👍

Mantamatt1, 2021-07-24:
> This is the image used for the bootloader - it's basically just tiled to fit the fill screen. So you should be able to use it to make your own 480×272 px PNG image, and shove it into [...]

Themes/edgetx/background.png

pfeerick, 2021-07-25:
Argh! sorry, yes, `/THEMES`, not `/IMAGES` ;)

closed 2021-07-25 by pfeerick