EdgeTX/edgetx #487: No access protocol selectable for X10 with ISRM upgrade

Skippydi, 2021-07-22:
With my X10s with the ISRM upgrade, I cannot select an ACCESS protocol in binding mode.

**General information**
 - EdgeTX version ( 2.4.0  and  98b1cd3d)
 - Hardware (Frsky Horus X10s with ISRM Upgrade)

raphaelcoeffic, 2021-07-22:
> With my X10s with the ISRM upgrade, I cannot select an ACCESS protocol in binding mode.
> 
> **General information**
> 
> * EdgeTX version ( 2.4.0  and  [98b1cd3](https://github.com/EdgeTX/edgetx/commit/98b1cd3dc814fe76a44f8cca2fb14108db6bda0b))
> * Hardware (Frsky Horus X10s with ISRM Upgrade)

@Skippydi could you please list your `cmake` options used during compilation? This configuration is not available by default, it case that was what you were trying.

Skippydi, 2021-07-22:
I didn't compile the firmware, but rather copied it to the X10s using the
Edge TX Flasher.

raphaelcoeffic, 2021-07-22:
> I didn't compile the firmware, but rather copied it to the X10s using the Edge TX Flasher. Raphael Coeffic ***@***.***> schrieb am Do. 22. Juli 2021 um 21:51:
> […](#)

Ah ok! So this normal: it is still assuming you have the normal iXJT inside.

Skippydi, 2021-07-22:
No, sorry. I have changed the IXJT Modul to X10 ISRM upgrade Modul Kit.

raphaelcoeffic, 2021-07-22:
> No, sorry. I have changed the IXJT Modul to X10 ISRM upgrade Modul Kit.

I see. As of now, this requires self-compiling, until we have an online platform that will allow us to support compilation options (or remove that compilation option and exchange it against a run-time option).

rotorman, 2021-07-22:
@Skippydi have a look at https://github.com/EdgeTX/edgetx/wiki/Building-and-debugging-EdgeTX to get your build environment setup and tips for self compiling.

Skippydi, 2021-07-22:
Wow great!! Thanks a lot for this info 👍🏽👍🏽👍🏽

pfeerick, 2021-07-26:
Closing as this is expected behavior. Feel free to continue to comment, or get help via [Discord](https://discord.gg/wF9wUKnZ6H).

closed 2021-07-26 by pfeerick