EdgeTX/edgetx #488: Help file missing?

fernfort, 2021-07-22:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

fernfort, 2021-07-22:
Are we going to have on Edge the function to call the model help file similar like OTX?
Used that on same models as quick reminder which switches do what, etc...very useful.
The text file is on MODELS and so far long press on enter...no joy.

Hardware: RadiMaster TX16S.

eshifri, 2021-07-22:
Do you enable it (model notes) in the model setup?
If I understand correctly what are you talking about the feature is implemented.
The file should be displayed every time the model is loaded. At least in RC4.

fernfort, 2021-07-22:
eshifri,
OTX displays help files done by user, long press enter will popup window where you select VIEW NOTES.
Check my post **Help file missing?** [https://www.rcgroups.com/forums/showthread.php?3916381-Official-EdgeTX-Version-2-4-0-Endeavour-Discussion-Thread/page18](url)

eshifri, 2021-07-22:
@fernfort 
Oh, now I understand - I have seen it there.
Currently EdgeTX displays modelNotes when you change the model (or start the TX).
You want to be able to call this file any time?

fernfort, 2021-07-22:
@eshifri ,
That is correct, I ask to have my model notes displayed on demand (long press enter or other way chosen by developers).
I usual use this before flight so, call VIEW NOTES can be implemented to fit actual touch philosophy.
Thanks for your attention, hop this will be available when dev´s have some spare time to fix it.

eshifri, 2021-07-22:
It is not difficult to implement this with long press/long touch on the "select model" icon in the carousel.
We just need to bring in the libopenui changes for long touch detection.

pfeerick, 2021-07-24:
That, or add another icon if the model notes are detected. At the moment you can view the notes view the browser, but it's certainly not as easy as a direct menu icon/shortcut. I would probably prefer an separate icon so we don't hide too many things behind perhaps not so intutitive multi-function icons.

eshifri, 2021-07-24:
Yes, another icon is much easier to implement, as I have found out after messing with long press for two evenings. :-)

closed 2021-07-31 by raphaelcoeffic

fernfort, 2021-08-01:
> Closed #488 <https://github.com/EdgeTX/edgetx/issues/488> via 645afc1

Hi  Raphael,

Where now the Model notes .txt file should be placed on SD card?

On a flasher I selected edgetx-firmware-main and selected latest
august-01T09:40:08Z
Is that correct firmware to get model notes?

Thanks for fixing Model notes!
Fernando Fortunato
Mobile: +351 962 377 536
***@***.***

pfeerick, 2021-08-01:
Should be fine as was merged two days ago. SO will also be in the last nightly or two as well. There is a later update which also updates the icon used to be more 'note-like' which is either in the last nightly or will be in today's. 

Model notes files go in the `/MODELS` directory like with OTX, and is subject to the same naming rules AFAIK.

commit 645afc11 (2021-07-31): Add modelNotes icon to the carousel if notes exist

Resolves #488