EdgeTX/edgetx #491: Special Functions Repeat input field: no scrolling in both directions

jogybaer0815, 2021-07-23:
TX16S - Endeavour (2.4.0) 09a6580

Input field 'Repeat' is showing little bug: 
If i am scrolling down to 1x from 60s i am able to reverse direction an vice versa. 
But one step further to '!1X' there is only one way back 60s and lower (same from 60s no way down/round)
Is that intended?

NemesisXB, 2021-07-23:
> TX16S - Endeavour (2.4.0) [09a6580](https://github.com/EdgeTX/edgetx/commit/09a65807ecee1b8b4b583b795931f8bc6690d437)
>
> Input field 'Repeat' is showing little bug:
> If i am scrolling down to 1x from [...]

Don't think it is intended. I also saw this, just never got to opening an issue for this.

jogybaer0815, 2021-07-23:
Of course not a big issue but a little annoing :)

pfeerick, 2021-08-01:
More specifically - this relates to the `Play Sound` or `Play Track` Special Functions - as they have the repeat boxes. 

Reproducable on the TX16S with 55098b66. Works properly on the X9D+ and T-Lite, so is a bounding issue on the colourscreens.

closed 2022-02-06 by pfeerick